esp8266/Arduino #3453: NodeMcu ESP8266 E12 Development board seems to auto sleep

wmweis62, 2017-07-23:
I am using a ESP8266 E12 development board to drive 4 relays for a project I am working on. I considered using the development board in my production solution, but it appears to go to sleep after a few hours. I have not figured out exactly what the cause is, but it seems I have to power cycle it to get it back online. I suppose I could be in an over temp situation. I am going to leave it powered on with the relays disconnected. Since I am new to this IoT stuff, are these development boards only intended for development purposes? (ie: Do I need to get a non development ESP8266 E12 chip for the production implementation?)

### Basic Info

Hardware
Makerfocus ESP8266 Module ESP-12E NodeMcu LUA WiFi Internet New Version Development board.

JBtek 4 channel DC 5v Relay module for Arduino Raspberry Pi DSP AVR PIC ARM

AMS1117-3.3v Power Supply Module Input 4.5-12v Output 3.3v

SNANSHI 5v 5A 25W LED Switching Power Supply Transformer 110 to DC5v output for LED Strip Light

Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu 1.0 (ESP 12-E module)
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			
Flash Frequency:	
Upload Using:		On board Serial
Reset Method:		

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

JamesGKent, 2017-07-24:
instead of completely changing the body of the post, you should be able to mark it closed, so it no longer appears as an active issue.

wmweis62, 2017-07-24:
Got it, thanks

wmweis62, 2017-07-24:
James,

I do not see how to mark it closed.  ( I will be offline for the next several hours)

JamesGKent, 2017-07-24:
you will probably need to log in on the website rather than replying via email, and there should be a button near the reply box to close the issue.

wmweis62, 2017-07-24:
Hi James,

I logged in and cannot see a button to close the posting near the Reply button. What am I missing?

[cid:image001.jpg@01D3048D.AA93A920]

JamesGKent, 2017-07-25:
you should see something like this:

[image]
note the grey button next to the comment button labeled "close issue"

wmweis62, 2017-07-25:
Bad 8266 chip

wmweis62, 2017-07-25:
Ah, it is closed now. Sorry, I had a similar post on the 8266 forum and I made the assumption you were telling me how to “close the Issue” on that forum.  I should have looked closer that you were talking about the posting on Github

[image]<https://user-images.githubusercontent.com/11315221/28562009-91dc5d9e-7118-11e7-8b9a-c178bdb4540a.png>

closed 2017-07-25 by wmweis62