esp8266/Arduino #3457: getting the esp8266 to connect to a hidden ssid

emilevs, 2017-07-24:
Please fill the info fields, it helps to get you faster support ;)

### Basic Infos

#### Hardware
Hardware:	        wemos d1 mini
Core Version:      	?2.1.0-rc2?

### Description

I am trying to get my wemos d1 mini to connect to a hidden ssid, it is a network with WPA2-PSK security. I have tried to connect to this network with a few of the example scripts and it is not working. I have looked into it and found this forum post https://github.com/esp8266/Arduino/issues/2390, there somebody came to the conclusion that his ESP8266 was set to n-only mode and it needed to be set to b/g/n mode. I am not sure what he meant by this. I did some research and apparently it has something to do with radio types but I still don't really understand it because it seems that you can only set it to n b or g not all 3.

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		arduino ide
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

#include <ESP8266WiFi.h>

void setup()
{
  Serial.begin(115200);
  Serial.println();
  WiFi.begin("ssid", "password");

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(WiFi.status());
  }
  Serial.println();

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {}

### Debug Messages

running this script will print WiFi.status() and that results in a lot of 6 (WL_DISCONNECTED)  sometimes a 4 (WL_CONNECT_FAILED) and sometimes a 1 (WL_NO_SSID_AVAIL)

5chufti, 2017-08-03:
it was not the esp that needed change of setting, it was the router ...

devyte, 2017-09-26:
@5chufti is your use case addressed with your router change? If so, then please close this.
If not, please try PR #3215 to see if it addresses it. There is a known issue when trying to connect to an AP with hidden beacon, and the changelist for SDK 2.1.0 mentions a fix.

5chufti, 2017-09-26:
Hi devyte, 
I didn't have the issue, only tried to explain #2390 to emilevs

devyte, 2017-09-26:
Sorry, feedback request was for @emilevs

emilevs, 2017-09-27:
I recenlty found out that they mistyped my ssid when I gave it to them, so that pretty much explains it

devyte, 2017-09-27:
Ok, then closing as resolved.

closed 2017-09-27 by devyte