esp8266/Arduino #3459: Cannot access my wifi router because there are two wifi with same SSID. How to write fuction wifi.begin()?

trungkiendt9, 2017-07-25:
When wifi scan, it see 2 wifi router:
1. "Netis", "-30dB", BSSID:.....
2. "Netis", "-78dB", BSSID:.....

Wifi.begin(Netis, *******) cannot connectted?.

Pls help!

JamesGKent, 2017-07-26:
i believe there is another `Wifi.begin` overload that takes a bssid parameter, you could check the docs and try adding the bssid for the one you want.

trungkiendt9, 2017-07-27:
OK. Thanks. Now it resolved.
I found the syntax: 
begin(char* ssid, char *passphrase = NULL, int32_t channel = 0, const uint8_t* bssid = NULL, bool connect = true);
I re-write:

const char* my_ssid     = "netis";
const char* my_password = "password";
unsigned char my_bssid[18] = { 0x.. 0x.., 0x38, 0x24, 0xba, 0x..};
It connected!.

JamesGKent, 2017-07-27:
if it worked then please close this issue.

Thanks

closed 2017-07-27 by trungkiendt9