esp8266/Arduino #3460: Some trouble to read GPS and IMU data

bartleouf, 2017-07-26:
Hi,

I am using a _Huzzah Feather ESP8826_ from Adafruit plugged into a _TFT FeatherWing_ screen.
I have connected an IMU _BNO055_ and a _Neo-6M_ GPS module (from Adafruit too).
The TFT works just fine when only the IMU is called by the sketch, or only the GPS, but not both at the same time (the outputs then are wrong). It bugs when I uncomment the tft.begin() in the Setup, even without a tft.print() in the Loop.

Could it be a matter of ressources? I tried to reduce the pace at which my ESP harvests the data, and put some yield() here and there, but it doesn’t seem to be of any help.
I am using SoftwareSerial() and Wire.pins(), maybe some weird conflict between them?

If you have any idea about it...

Thanks

```
#include <SoftwareSerial.h>
#include <TinyGPS.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <Adafruit_BNO055.h>

#define STMPE_CS 16
#define TFT_CS   0
#define TFT_DC   15
#define BNO055_SAMPLERATE_DELAY_MS (500)
#define rx_pin 13
#define tx_pin 12

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
Adafruit_BNO055 bno = Adafruit_BNO055();

TinyGPS gps;
SoftwareSerial rxtx = SoftwareSerial(rx_pin, tx_pin);

struct timelatilongi {
  int annee;
  byte mois, jour, heure, minuttte, seconde, centieme;   
  float lati, longi;
};

// ------------------------------ Setup & loop ------------------------------

void setup(void) {

  wdt_enable(WDTO_1S);
  wdt_reset();
  pinMode(rx_pin, INPUT);
  pinMode(tx_pin, OUTPUT);
  Serial.begin(115200);
  rxtx.begin(9600);
  //tft.begin();
  yield();
  Serial.begin(115200);
  starting_bno();
  yield();

}

void loop(void) {
 
  wdt_enable(WDTO_1S);
  wdt_reset();
  timelatilongi position=search_gps();
  float latitude=position.lati, longitude=position.longi;
  float heading=direction();
  Serial.print("latitude : "); Serial.println(latitude);
  Serial.print("longitude : "); Serial.println(longitude);
  Serial.print("heading : "); Serial.println(heading);
  Serial.println();
  //tft_display(latitude, longitude, heading);
  yield();
}

// ------------------------------ Functions ------------------------------

void starting_bno() {
  if(!bno.begin())  {
    Serial.print("Ooops, no BNO055 detected ... Check your wiring or I2C ADDR!");
    while(1);
  }
  delay(1000);
  bno.setExtCrystalUse(true);
}

struct timelatilongi search_gps() {
  int year;
  byte month, day, hour, minute, second, hundr;
  bool newData = false;
  unsigned long chars; unsigned short sentences, failed;
  float flat; float flon; unsigned long age;
  struct timelatilongi coord;

  // For one second we parse GPS data and report some key values
  for (unsigned long start = millis(); millis() - start < 1000;) {
    while (rxtx.available()) {
      char c = rxtx.read();
      if (gps.encode(c))
        newData = true;
    }
  }
  if (newData) {
    gps.f_get_position(&flat, &flon, &age);
    gps.crack_datetime(&year,&month,&day,&hour,&minute,&second,&hundr);
    coord.lati=flat;
    coord.longi=flon;
    return coord;
  }
  gps.stats(&chars, &sentences, &failed);
  if (chars == 0) { Serial.println(F("** No characters received from GPS: check wiring **")); }
  delay(1000);
}

float direction() {
  sensors_event_t event;
  bno.getEvent(&event);
  yield();
  float cap=event.orientation.x;
  delay(1000);
  return(cap);
}

void tft_display(float latit, float longit, float head) {
  tft.fillScreen(ILI9341_WHITE);
  tft.setTextColor(ILI9341_BLACK);
  tft.print("Latitude : ");  tft.println(latit,6);
  tft.print("Longitude : ");  tft.println(longit,6);
  tft.print("Heading : ");  tft.println(head,0);
}
```

bbx10, 2017-07-26:
Good question for Adafruit's forum. But the first thing to try is to move the softserial pins somewhere else.

#define rx_pin 13
#define tx_pin 12

Pins 12,13,14 (SPI MISO, MOSI, CLK) are used by the display/SD card.

Aureil, 2017-07-28:
Yes I posted a message on Adafruit’s forum, but no answer so far.
Thanks bbx10 for your remark, I had tried several pin combinations and got mixed up. So now rx is 5 and tx is 4, and the GPS is working as long as I don’t call starting_bno(); 
I tried to isolate the IMU call inside an interrupt, triggered from the loop() . But this kind of operation seems to take too long for an interrupt, especially since I have to intiialize the IMU everytime I call it. 
Is there any kind of interrupt that would allow a longer break in the loop course?

bbx10, 2017-08-02:
The default for i2c devices such as the BNO is to used pins 4 and 5 so software serial cannot use the same pins. Leave the BNO as is and try pins 0 and 2 for softserial/GPS.

I do not think interrupts will help for this situation.

Aureil, 2017-08-04:
Yes, it looks like I have to forget about the SD feature or some other bad solution. But actually there are a few GPS modules that run on _sda/scl_. I am gonna try with the Venus from Sparkfun.
Thanks for your help anyway!

closed 2017-12-28 by igrr

OldRavenNL, 2019-06-26:
Did you have any luck in getting this to run? I have a Huzzah and TFT featherwing, and would also like to get this to work.
regarding bbx10's answer..  I don't mind sacrificeing the touch function of the screen. I need the SD to log the GPS data.