esp8266/Arduino #3464: Problems when uploading sketches to the Wemos d1 mini on arduino IDE

daecepi, 2017-07-26:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos
Each time i connect my ---Wemos d1 mini--- to upload a sketch arduino IDE shows me this DEBUG MESSAGE:

error: cannot access /dev/ttyUSB0

error: espcomm_open failed
error: espcomm_upload_mem failed
the selected serial port 
does not exist or your board is not connected

-->The problem started when i uploaded this code:

//--------------SCRIPT---------------
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer server(80);

//Mis credenciales de la red a crear
char* mySsid="CarritoWiFi";
char* myPassword="SuperCarroFantastico";

void setup() 
{
   delay(1000);
  Serial.begin(115200);

  
  Serial.println("Configurando Access Point...");

  //WiFi.mode(WIFI_AP);
  WiFi.softAP(mySsid,myPassword);
  IPAddress ip = WiFi.softAPIP();

  Serial.println("IP:");
  Serial.print(ip);

  server.on("/movement", controlMovement);
  server.begin();
}

void loop(){
  server.handleClient();
}

void controlMovement(){
  String pass = server.arg("p");
  if(pass=="2244"){
    String forward = server.arg("f");
    String backward = server.arg("b");
    String right = server.arg("r");
    String left = server.arg("l");
    Serial.println(forward+backward+right+left);
    server.send(204,"");
  }
}
------END OF SCRIPT------
When finished uploading i pressed the reset button because the baud speed in the serial monitor was different to the one i put, and the message printing the IP seemed to have happened, so i heard that pressing the reset button would just restart the script; after severous number of tries decidered to see if it showed anything in the serial monitor (it does it all the time connected to the computer):

-----MESSAGE ON THE SERIAL MONITOR-----
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset

load 0x4010f000, len 1384, room 16 

tail 8

chksum 0x2d

csum 0x2d

v09f0c112

~ld
-----END OF MESSAGE ON THE SERIAL MONITOR-----

Thanks in advance  😄

devyte, 2017-07-27:
> Troubleshooting/stack_dump.md
> Troubleshooting/debugging.md

The error message clearly says that ttyUSB0 is not accessible. Either it
changed to something else, and you need to go to the IDE menu and set the
serial port again, or there is some other problem with your serial port (I
had to reboot my Linux pc once to get it back).
Make sure that /dev/ttyUSBsomenumber actually exists, and that you have
permissions set correctly to access it. If you don't know how to do that,
please google it or ask for help on a forum (not here, this is an issue
tracker for the core libs).

daecepi, 2017-07-29:
Sorry for any inconvenient, the problem was with the VM adapter.

closed 2017-07-29 by daecepi