esp8266/Arduino #3465: ESP-12E cannot get connection after several times reset

zuzy, 2017-07-27:
I restore the ssid and password in flash. After power on, i call wifimulti to connect the router. 
When i power off the module, the list of connection in router tells the module is still online. 
Then i power on the module, the module connect successfully and the list of connection get refreshed.
But after i reset the module several times, the module cannot get connection until i reset the router.
Is there any  configure of the module to solve the problem?

rolpal, 2017-08-06:
Try to read the router manual and see if there is maximum limit of connections to the router.
It might take some time for the router to "age out" the "old" connections and you need to reset it to do it.
Many low cost routers have a limited number of entries in their station table.

How many times are several times, and how often?

zuzy, 2017-08-16:
I use flash to store the ssid & password. I read the flash as power on, then wifi.begin to connect the router. If the board cannot get connection after 30 secs, i start to smartconfig. 
I found that sometime the board cannot get connection with the function WiFi.begin(ssid, pass), but it can connect to the same router and get the old ip when smartconfig.

devyte, 2019-11-10:
This issue is old. A whole lot has changed since it was reported, including several fixes that are relevant.
In addition, the report is not compliant with the current issue POLICY or ISSUE_TEMPLATE.
Closing.
If you think the problem is still relevant, please open a new issue and follow the template instructions.

closed 2019-11-10 by devyte