esp8266/Arduino #3467: HTTP server slow on simultaneous requests

tttapa, 2017-07-27:
### Basic Infos

#### Hardware
Hardware:			WeMos D1 mini
Core Version:      	2.3.0

### Description

When sending multiple HTTP requests at the same time to the web server running on the ESP, the server responds very slowly (up to 23 seconds) and some requests even fail (because the ESP sends an empty response). 
If I change my client code so that it waits for one request to finish before sending the next request, these problems do not occur (all requests finish in less than 40ms).

#### All requests at once:
[image]
#### One request after the other:
[image]
Blue: Content download
Green: Time to first byte (waiting)
Grey: Stalled

Sometimes, _output?output=5_ fails, sometimes it doesn't. Not sure why.
### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4M (1M SPIFFS)
CPU Frequency:	80MHz
Upload Using:		Serial (921600 baud)

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <FS.h>

ESP8266WebServer server = ESP8266WebServer(80);       // create a web server on port 80

const char* WiFiHostname = "esp8266";      // Host name on the network

const char* ssid     = "SSID";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "PASSPHRASE";     // The password of the Wi-Fi network

const uint8_t outputs[] = { 2, 4, 5, 12, 13, 14, 15, 16 };
const size_t  nb_outputs = sizeof(outputs);

/*__________________________________________________________SETUP__________________________________________________________*/

void setup() {

  Serial.begin(115200);        // Start the Serial communication to send messages to the computer
  delay(10);
  Serial.println('\n');

  startOutputs();              // Set all LED pins to outputs

  startWiFi();                 // Start a Wi-Fi access point, and try to connect to some given access points. Then wait for either an AP or STA connection

  startSPIFFS();               // Start the SPIFFS and list all contents

  startServer();               // Start a HTTP server with a file read handler and an upload handler

}

/*__________________________________________________________LOOP__________________________________________________________*/

void loop() {
  server.handleClient();      // run the server
}

/*__________________________________________________________SETUP_FUNCTIONS__________________________________________________________*/
[... 285 lines cut ...]
GET 	Output: 3
[G 3]	12055	Start sending
[G 3]	12068	Finished sending	(13)
GET 	Output: 5
[G 5]	12073	Start sending
[G 5]	12081	Finished sending	(8)
GET 	Output: 4
[G 4]	12086	Start sending
[G 4]	12097	Finished sending	(11)
GET 	Output: 6
[G 6]	12102	Start sending
[G 6]	12110	Finished sending	(8)
GET 	Output: 7
[G 7]	12114	Start sending
[G 7]	12129	Finished sending	(15)
```
I know it's a lot of code, but many parts are the same as (or very close to) the [FSBrowser example](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/FSBrowser).

The JavaScript code first asks the ESP how many outputs it has (GET request to `/nb_outputs`). Then it creates the right number of buttons (one for each output) and asks the ESP what the state of each output is (GET request to `/output?output=i`, where i is the number of each button). Finally, it changes the state of the buttons on the webpage according to the response it got from the ESP. 
By clicking a button, it sends the new state to the ESP (POST request to `/outputs` with a body containing urlencoded formdata `output=i&state=x` where x is either '0' or '1').

I really don't know why the first request takes so long (10s according to the ESP, 23s according to Chrome) and why the the other requests all finish after 12s (according to both the ESP and Chrome).

Thanks in advance, 
Pieter

JamesGKent, 2017-08-07:
the webserver can only handle a single request at a time, so receiving multiple requests at once (so more come while dealing with the first) drags the flow of execution away from your code to handle the background wifi related tasks, this will slow down the execution and use more resources, it's possible you are running out of ram/stack causing some of your requests to fail.
i would suggest that issuing a separate request for each output is bad design, instead i would make it so you have a single request to get all output states at the same time as you request the number of outputs.

Just a thought.

however if you are insistent on this design, look into the async libraries for esp, specifically asyncwebserver

tttapa, 2017-08-07:
Thank you for the reply.
In the final version, I send everything in a single response, but I just found it strange that it fails so easily. I thought that there may be a problem with the actual code I used in the server handlers.
I'll definitely look into the async libraries, looks very interesting. 

Thanks again,
Pieter

devyte, 2017-08-07:
A note here, from my experience a few months backago.
The async webserver has many good things, but there is one downside: you
can't call yield() or delay() from an endpoint handler, because it doesn't
run in the loop.
This means that, if your handler takes a long processing time, you'll run
into wifi stack corruption.
In contrast, the standard webserver runs in the loop, so if your handler
takes a long time, you can call yield() in key places to allow the wifi
stack to process before the handler is done.

There are ways around this limitation in the async webserver, but they are
nontrivial and make for very complex code.

Personally, after trying out both, I'm sticking to the standard webserver,
but of course that's just my own choice.

JamesGKent, 2017-08-08:
fair enough, one thing i can add is that using the `String` type and joining them together can be a big cause of the slowdown, along with memory fragmentation, for my code i adapted it to using a big buffer assigned at the start of the handler, and then `strcpy` data into my buffer, i found this to be orders of magnitude faster than the strings, and a lot more stable too, you could look at that instead?

perote2000, 2017-08-14:
Hi, could you write a small example of code about using buffer insteaf of strings. Are you telling about client.print for instance.. ?

JamesGKent, 2017-08-14:
so not a complete example, but i add a variable like this:
#define RETBUF_SIZE 1420
char retbuf[RETBUF_SIZE];

then in the function where you receive a new request start with:
memset(retbuf, 0, RETBUF_SIZE); // empty the reply buffer
uint16_t retlength; // to keep track of where in buffer we have written to

then to add new data to the buffer:
retlength += sprintf(&retbuf[retlength], "new string stuff");

note that if you want to add numbers you'll have to do the conversions yourself

this could of course be encapsulated in a class much like how the arduino string class works, but the difference is the memory gets assigned once in a contiguous block rather than little chunks all over the place.

tttapa, 2017-08-14:
Do you mean the Strings used in the actual server class that comes with the ESP8266 Arduino Core?

JamesGKent, 2017-08-14:
yes, the built in Arduino String type that comes with the core allocates more memory when you add them together or add sting literals etc into them. allocating memory like that is slow, and also means that the figure the IDE gives you for memory used becomes inaccurate, and causes memory fragmentation. by assigning a big buffer for any built responses you get a single allocation (faster) with a single contiguous block (no fragmentation) and because i defined mine as global its a known amount of memory used, meaning i have a better idea how much i'm using overall from the output of the compiler.

however rereading your code you only seem to use the string type in your file handler rather than for your built responses, in which case you are probably causing memory exhaustion with your flood of connections

tttapa, 2017-08-14:
I'm well aware of the disadvantages of String objects, what I meant was that the server class itself uses Strings rather than null terminated char arrays for things like HTTP headers and arguments:
[ESP8266WebServer.h](https://github.com/esp8266/Arduino/blob/72ed29ad5f18aac3d26037e4a8ff14cc523eb7a1/libraries/ESP8266WebServer/src/ESP8266WebServer.h#L97)

everslick, 2017-11-06:
FYI: The String class has the reserve() method. If you know how big your html code is going to be, just reserve as much as needed before adding substrings to it. This will avoid the whole fragmentation/reallocation/memcopy problem and you get all benefits from the String class.

```
String str;

str.reserve(1024);
str = F("foo");
str += F("bar");
...
```

PotatoX, 2017-12-02:
Also consider compartmentalizing your code a bit and using scrip synchronization? This worked for me to keep the browser from blasting my ESP webserver with a bunch of simult loads.

https://www.w3schools.com/tags/att_script_defer.asp
-or-
https://www.w3schools.com/tags/att_script_async.asp

devyte, 2019-11-20:
This issue is old. A lot has changed since it was reported, including a long list of relevant fixes.
Closing due to age.
If you think the problem is still relevant, please open a new issue, follow the template instructions, and add details like in this report.

closed 2019-11-20 by devyte