esp8266/Arduino #3470: Esptool, baudrate and compiled binary caching

dimitre, 2017-07-28:
I've been writing the same firmware to flash multiple ESPs (ESP-01) and I've been using esptool 1.2 to upload it faster to the ESPS

I've been able to achieve much higher baudrates directly on esptool 1.2. 3000000 vs 230400 in IDE
```
esptool.py --baud 3000000 --port /dev/cu.usbserial-00001014 write_flash --flash_mode dio --flash_size detect 0x0 /Volumes/ofw/TEENSYATUAL/_ESP/ESTILHACO3/ESTILHACO3.ino.generic.bin 
```

And I've noticed in Arduino IDE it recompiles everything even if the code is untouched.

It would be great if the code could be cached as in Teensyduino and if we could achieve higher baudrates inside IDE.

Thank you

igrr, 2017-12-27:
> And I've noticed in Arduino IDE it recompiles everything even if the code is untouched.

Just checked on OS X with Arduino 1.8.2, and this doesn't seem to be the case in latest git version. Only the static library is created on each build, but object files are not rebuilt.

Regarding upload baud rate, please feel free to open an enhancement request in https://github.com/igrr/esptool-ck.

closed 2017-12-27 by igrr

dimitre, 2018-02-23:
Hello @igrr in fact I was talking about the baudrate itself. I can achieve very high baudrates uploading using esptool directly from command line, but inside Arduino IDE it works only with slow baudrates. 
If you have the time make the comparsion itself using a similar command to the original one I've pasted: 
```
esptool.py --baud 3000000 --port /dev/cu.usbserial-00001014 write_flash --flash_mode dio --flash_size detect 0x0 /Volumes/ofw/TEENSYATUAL/_ESP/ESTILHACO3/ESTILHACO3.ino.generic.bin 
```
Thank you