esp8266/Arduino #3471: Problem with uploading the sketch

daecepi, 2017-07-29:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 Mini

### Description
A friend of mini borrowed me an Wemos D1 Mini when trying to upload the sketch to the board arduino IDE shows me  an error (specified in the debug messages section), i try with mine and uploads normally, for an assigment i need both functioning, but first i´m testing it with the blink example and doesn't work

¿Any ideas of what could be the problem? ¿May it need a hard reset or something?

I'm a bit new to boards and arduino, thanks in advance 😄 

### Settings in IDE

Module:  			Wemos D1 Mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			Programer: AVRISP.mkll
Flash Frequency:	No aparece entre las opciones
Upload Using:		SERIAL?
Reset Method:		Do not know what it is

### Sketch
---the blink example---

### Debug Messages

Archiving built core (caching) in: /tmp/arduino_cache_746057/core/core_esp8266_esp8266_d1_mini_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_7165ef2ba1fc0af9ea593fd0b652a5e2.a
Sketch uses 222117 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31492 bytes (38%) of dynamic memory, leaving 50428 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

rolpal, 2017-08-06:
This look like a typical HW fault where the device is not in flash mode..
Have you followed the wiring instructions for programming/flashing?

devyte, 2017-09-05:
I doubt you should be using AVRISP to program a Wemos D1 mini. 
This looks like a user error. Please research for an answer, or ask at esp8266.com/stackoverflow for assistance. I can vouch for the Wemos D1 mini not being broken, because I'm using it myself.
Closing.

closed 2017-09-05 by devyte