esp8266/Arduino #3474: issue regarding exception 9 in esp8266

anjushar, 2017-07-29:
Hello sir, i am getting exception 9 ...for a program. i referred many forums ..but i could not solve it.can you help me in solving. and code is as follows
i am publishing from one program
Publish message: starttime:18:29:18&field2=0.54&field3=0.00&field4=0.00

and trying to subscribe using another program and to diplay a difference time on lcd.....16x2
---when i remove lcd.print(data); in below code ,it is not showing exception 
---when i add its showing exception....
---i understood problem is related to wdt ...and program resetting again and again.please help me sir to solve.

[code.docx](https://github.com/esp8266/Arduino/files/1184967/code.docx)

devyte, 2017-07-29:
Which board? Which sdk? Core?

Off the top of my head:
-your reconnect scheme seems wrong to me. If a reconnect should ever be
attempted, the wdt will likely fire. You need to let the loop run, and make
reconnection attempts on each pass.
-you have a global char data[30], which is then shadowed by  a loca String
data.
-check the pin definition for your specific board. I see that you are using
D3, which is used for flash access on some boards. Try different pins.

devyte, 2017-09-05:
Closing due to lack of info and lack of response in over a month.

closed 2017-09-05 by devyte