esp8266/Arduino #3476: OTA: how to save user data on the second half

soheils2, 2017-07-29:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

hi there im using OTA update for my esp8266 
in this mode :
Flash layout
esp use only half the flash space with the firmware such that two partitions can be created. When the firmware is running out of partition 1 then partition 2 can be loaded with a new version and a restart can make the esp8266 run out of the new partition 2. At that point partition 1 can be updated, and so forth.

i was wondering to use another half of flash to save user data on it ; and it can keeps the information until next update.
can u help me with that?

devyte, 2017-07-29:
> Troubleshooting/stack_dump.md
> Troubleshooting/debugging.md

Hi,

The firmware doesn't use half of the flash. The currently used firmware
uses the first 512KB of flash, while the OTA update is written in the
second 512KB of flash (I'm over simplifying).
For a 4MB ESP module, that means you have about 3MB left of space.
If you want to store user data, then you should use the SPIFFS and store it
in files.
You can also use the EEPROM interface, which is a thin layer over the flash
I/O interface to mimic the Arduino EEPROM.
Or you can choose your own flash address space somewhere that isn't used,
and do direct read/write there.

However, if you use the EEPROM or direct flash, you will wear out those
sectors faster than others, until they fail, and your ESP will then be no
good.
In contrast, if you use SPIFFS and files, the file system does wear
levelling under the hood, so wear is spread out over the entire size of the
SPIFFS (you can choose either 1MB or 3MB for SPIFFS size).

Now, having said all that, this is not the right place to ask this kind of
question. This is an issue tracker, to track bugs and feature requests for
the core libs. That's why they're called "issues". Please close this and
ask for more info on a forum.

devyte, 2017-08-10:
@soheils2 Please close this, since it's a "how do I..." type question rather than an issue with the core libs.

closed 2017-09-05 by devyte