esp8266/Arduino #3479: Support Rising / Falling Edge for WakeUp light sleep

soheils2, 2017-07-30:
hi there 

The ESP8266 only wakes up when you use GPIO_PIN_INTR_LOLEVEL / GPIO_PIN_INTR_HILEVEL,
So for GPIO_PIN_INTR_LOLEVEL you would need to Pull the voltage down to 0V then to 3.3V up again.

i am trying to wake my esp by a gpio that defined as software serial.
and when the board on the other side is ON we always get HIGH (if highlevel int attached it goes on the bad loop) then i must chose low level
and would be the same when the board on the other side is OFF we always get LOW (if lowlevel int attached it goes on the bad loop) then i must chose high level int

The solution would be a using 
    GPIO_PIN_INTR_POSEDGE 
    GPIO_PIN_INTR_NEGEDGE 
    GPIO_PIN_INTR_ANYEDGE,
 But If I use this, nothing happens

I search into some codes and libraries and found this low level function at "gpio.c" :

``
/*
  only highlevel and lowlevel intr can use for wakeup
*/
void ICACHE_FLASH_ATTR
gpio_pin_wakeup_enable(uint32 i, GPIO_INT_TYPE intr_state)
{
    uint32 pin_reg;

    if ((intr_state == GPIO_PIN_INTR_LOLEVEL) || (intr_state == GPIO_PIN_INTR_HILEVEL)) {
        portENTER_CRITICAL();

        pin_reg = GPIO_REG_READ(GPIO_PIN_ADDR(i));
        pin_reg &= (~GPIO_PIN_INT_TYPE_MASK);
        pin_reg |= (intr_state << GPIO_PIN_INT_TYPE_LSB);
        pin_reg |= GPIO_PIN_WAKEUP_ENABLE_SET(GPIO_WAKEUP_ENABLE);
        GPIO_REG_WRITE(GPIO_PIN_ADDR(i), pin_reg);

        portEXIT_CRITICAL();
    }
}

``

from this library [https://github.com/espressif/ESP8266_MP3_DECODER](url)
and this part kinda make me upset:
``
/*
  only highlevel and lowlevel intr can use for wakeup
*/

``
now my question is: 
is there any way to wake up my ESP guy from light sleep using falling or rising  edges?

can anybody help me with that?

JamesGKent, 2017-07-31:
have you tried changing the function to this:
```
void ICACHE_FLASH_ATTR
gpio_pin_wakeup_enable(uint32 i, GPIO_INT_TYPE intr_state)
{
uint32 pin_reg;

if ((intr_state == GPIO_PIN_INTR_LOLEVEL) || (intr_state == GPIO_PIN_INTR_HILEVEL) || (intr_state == GPIO_PIN_INTR_POSEDGE) || (intr_state == GPIO_PIN_INTR_NEGEDGE) || (intr_state == GPIO_PIN_INTR_ANYEDGE)) {
    portENTER_CRITICAL();

    pin_reg = GPIO_REG_READ(GPIO_PIN_ADDR(i));
    pin_reg &= (~GPIO_PIN_INT_TYPE_MASK);
    pin_reg |= (intr_state << GPIO_PIN_INT_TYPE_LSB);
    pin_reg |= GPIO_PIN_WAKEUP_ENABLE_SET(GPIO_WAKEUP_ENABLE);
    GPIO_REG_WRITE(GPIO_PIN_ADDR(i), pin_reg);

    portEXIT_CRITICAL();
}
}
```

devyte, 2017-08-10:
@JamesGKent is that tested to work, or is it just a suggestion to try out?

JamesGKent, 2017-08-11:
a suggestion to try, i don't have my ESP to try it, but assuming those symbols you quoted are defined in the SDK/toolchain i don't see why this shouldn't work.

davescherler, 2017-08-22:
@soheils2 I'm also interested in waking up my ESP from light sleep with a Falling Edge interrupt. I'm currently using a DS3231 (real time clock module with alarm functions) to trigger a falling edge with an alarm every 20 seconds. I thought I could use this falling edge to wake up the ESP but alas. 

Have you made any progress? Also the link you added is broken (for me). Where exactly did you find that function? Were you able to implement the suggestion from @JamesGKent ?

davescherler, 2017-08-22:
@soheils2 Also I found this post on the Expressif forum that further validates the wakeup from light sleep can only be done by a LOW or HIGH level signal: 

> b).void gpio_pin_wakeup_enable(uint32 i, GPIO_INT_TYPE intr_state) : (parm i is the gpio number) To enable the wake-up-by-gpio function.In this situation, interrupt mode can just be set to GPIO_PIN_INTR_LOLEVEL, GPIO_PIN_INTR_HILEVEL.That means for gpio,only level interrupt can wake up the chip.Since the cpu will shut down , so the chip would not response to the uart or spi data, we can wake it up via gpio first. 

[(http://bbs.espressif.com/viewtopic.php?t=171)]

Man, this is really disappointing, and I'm not sure there are any workarounds for that. I should have researched that limitation before going all in on waking up from a falling edge. 

Would anyone happen to know any workarounds? Is there any way to use hardware to fake a LOW level out of a falling edge? Otherwise, I may have to rethink my whole project.

JamesGKent, 2017-08-22:
as a workaround you could add a non retriggerable monostable timer in line with your signal, what this means is a timer that only times once (mono rather than astable) and by being non retriggerable it will trigger once at the falling edge, but have to be taken high again in order to trigger a second time. so it would generate a short pulse which would trigger your wakeup, one way to do this would be with a 555 timer IC, but this seems a tad overkill to me, there are probably better ways to do this with less/smaller components.

note that this wouldn't solve the original question as it would stop the software serial from working correctly.

devyte, 2017-09-05:
@soheils2 @davescherler were you able to test the proposed suggestion?

davescherler, 2017-09-05:
@JamesGKent @devyte Thanks for following-up. Yes I have been able to test @JamesGKent suggestion but something isn't quite right. But I think I'm close. I went out a got a 555 timer IC (it may be overkill, yes, but it was the best starting place for me) and got it working as expected in monostable mode. I tested by powering the timer with the same voltage lines my ESP is running on; a 100uF cap and a 22k Ohm resistor; a momentary push button wired to the input pin and an LED running out of the timer's output. Based on this equation: time = 1.1*R*C, the LED should light up for 2.42 seconds (1.1*22,000*0.0001). And when I push the push button, the LED lights up for 2 and half seconds then goes off. This is exactly as I expect, since the output pin should output a HIGH signal for however long that capacitor will take to discharge, at which point the timer flips back. All good so far. 

The next thing I tested was substituting the push button for the output signal of my DS3231 RTC. In this setup, its my once-a-minute alarm that outputs a Falling Edge pulse which the timer can see and outputs a high signal. Again, the LED lights up as expected for 2 and a half seconds, then turns off. It does this every time the alarm goes off. This validates that the Falling Edge signal from the RTC can be seen by the timer and it will register a HIGH pulse. All good so far. 

So now I'm thinking, "great, I can see an Falling Edge and output a HIGH signal. All I need to do is make that HIGH signal shorter (by reducing the size of the cap/resistors) and send that signal back into my ESP to wake it up." (Note: I'm keeping my LED in line for now for testing). And this is where I'm stuck. After running about 60 seconds of a meaningless loop(), my ESP goes into Light Sleep. Once in sleep, and the next minute passes, the RTC alarm goes and the LED turns on. But the LED never turns off. It just stays on, the ESP doesn't register anything in the monitor, never runs my wakeupcb function, nothing. I didn't think I'd even be able to see the LED turn on (because the pulse would be so short) let alone it remain on forever. @JamesGKent does that mean I don't have the right value of capacitors and resistors? Does it mean that the timer is constantly registering a Falling Edge? Is it something to do with the fact that the ESP is asleep? This is really strange because I have a nearly identical project (still running) that uses a PIR sensor to wake up the ESP after it's in Light Sleep. So I know the ESP can wake up from a HIGH pulse. Any thoughts would be appreciated!

devyte, 2017-09-06:
@soheils2 @davescherler No, I didn't mean test with a 555 to wake up with level, but rather the code change proposed by @JamesGKent.
(about the referenced espressif post, keep in mind that many answers provided by espressif have been found to be incomplete before)

@JamesGKent could you please elaborate on where to try that? I thought it was a core function, but I just now realized that you seem to be talking about an SDK function.

JamesGKent, 2017-09-06:
i replied naively assuming that it was a core function, i didn't look too deeply.

i did try overriding the function within a sketch:
extern "C" {
#include "c_types.h"
#include "gpio.h"
}

void ICACHE_FLASH_ATTR gpio_pin_wakeup_enable(uint32 i, GPIO_INT_TYPE intr_state)
{
    uint32 pin_reg;

    if ((intr_state == GPIO_PIN_INTR_LOLEVEL) || (intr_state == GPIO_PIN_INTR_HILEVEL)) {
//        portENTER_CRITICAL();
		vPortEnterCritical();
		
        pin_reg = GPIO_REG_READ(GPIO_PIN_ADDR(i));
        pin_reg &= (~GPIO_PIN_INT_TYPE_MASK);
        pin_reg |= (intr_state << GPIO_PIN_INT_TYPE_LSB);
        pin_reg |= GPIO_PIN_WAKEUP_ENABLE_SET(GPIO_WAKEUP_ENABLE);
        GPIO_REG_WRITE(GPIO_PIN_ADDR(i), pin_reg);

//        portEXIT_CRITICAL();
		vPortExitCritical();
    }
}

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

however on further research the portENTER/portEXIT should defined as macros however they appear to be connected to freertos, not the non os SDK, so i'm not sure if this would ever work as neither the macros or the required functions are defined anywhere.

and i don't currently have any hardware to be able to do anything useful with in order to test

Tech-TX, 2020-02-01:
The latest RTOS API Reference shows that only LOW_LEVEL and HIGH_LEVEL interrupts work, so I'm reasonably sure it's a limitation of the CPU and not the SDK.

devyte, 2020-02-02:
EDGE trigger isn't supported. Unless/until someone comes up with a working examples that proves otherwise, that's the official stance by Espressif.
Closing.

closed 2020-02-02 by devyte