esp8266/Arduino #3483: DateTime.h missing

probonopd, 2017-07-31:
`DateTime.h` is normally available in Arduino (for AVR at least) but is missing fir esp8266/Arduino.

An implementation seems to be available at https://github.com/jeremymlong/ESP8266-UPnP - could it be integrated into esp8266/Arduino?

probonopd, 2017-08-12:
ping

igrr, 2017-08-14:
Do you mean that you want DateTime to be added as a part of the core?
I don't see DateTime class in the AVR core here:
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/cores/arduino

probonopd, 2017-08-14:
Nvermind, you are correct.

> This library has been superseded by a newer version that is available [here](https://www.pjrc.com/teensy/td_libs_Time.html)

https://playground.arduino.cc/Code/DateTime

closed 2017-08-14 by probonopd