esp8266/Arduino #3489: Connection to static IP fails

bbkiwi, 2017-08-02:
### Description
In the attached sketch, having a small delay at the beginning of setup() seems to cause connection to a static IP to fail. 

The code below produces the error: it fails to connect. I expected it connect quickly using static IP, but it usually fails to connect at all or takes a very long time. If I try a new static IP, sometimes it seems to connect once, but after resetting or even recompiling and loading fails from then on. 

In the example,  If either the delay(500); or WiFi.config(staticIP, gateway, subnet); is commented out
the problem goes away. Connection to the static IP is very fast or connection to the assigned IP as usual. 

(I looked at issue #128 but the problem is there with either order of config and begin)
Of course the solution is not to put a delay, but it is a puzzle why this happens :-)

### Settings in IDE
Module:  			NodeMCUv1.0 (ESP 12E module)
Flash Size: 		4M(3M spiff)
CPU Frequency:		80Mhz
Upload Using:		SERIAL 256000

 ```
 // Bug with connecting with static IP
 // Fails (except perhaps on first use of specified IP) to connects
//       (rarely will connect after very long time 30 secs)
// Removing delay fixes and connection to specified IP is very fast (1 dot)
// If comment out WiFi.config works fine (the delay has not effect), however
// Connection time is longer as connects via DHCP
// Tested on nodemcu v1.0 IDE 1.8.1 ESP8266 2.3.0

#include <ESP8266WiFi.h>

const char* ssid = "Test Station";
const char* password = "testpassword";
IPAddress staticIP(192,168,1,229);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

void setup() {
  Serial.begin(115200);
  //BUG? with delay below, fails to connect to specified IP (except perhaps on first use)
  delay(500); //problem happens with delays of 150 or more
  WiFi.config(staticIP, gateway, subnet);
  WiFi.begin(ssid, password);
 
  Serial.println();
  Serial.print("Connecting");
  while(WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("WiFi Connected, IP:");
  Serial.println(WiFi.localIP());
}

void loop() {
}
```

hazza-ob, 2017-08-02:
Have you tried testing this with other micro controllers? or even running it on the ESP module itself? May be due to the timers within the ATMEGA 328...

Although this is just theory

nbenm, 2017-09-30:
@bbkiwi I have similar problem.
I didn't see that running WiFi.config at the very beginning of setup would work.
As for you, it works fine with dhcp.
Connection is also very fast. I have no dot at all after first run.
I'm using same conf with ESP-12F board. Tried also 1M spiff.
Did you solve the issue ? If so, how ?
Thanks

bbkiwi, 2017-10-01:
I have not solved it. I have been programming using nonos directly and just 
wanted to see
if the similar problem occurred using Arduino ide (which it did hence me raising 
the issue).

Once it starts refusing to connect, if I reset my router,  it will connect again 
to specified IP.

lrmoreno007, 2018-01-08:
Try this:

```
 // Bug with connecting with static IP
 // Fails (except perhaps on first use of specified IP) to connects
 //       (rarely will connect after very long time 30 secs)
 // Removing delay fixes and connection to specified IP is very fast (1 dot)
 // If comment out WiFi.config works fine (the delay has not effect), however
 // Connection time is longer as connects via DHCP
 // Tested on nodemcu v1.0 IDE 1.8.1 ESP8266 2.3.0
 
 #include <ESP8266WiFi.h>
 
 const char* ssid = "Test Station";
 const char* password = "testpassword";
 IPAddress staticIP(192,168,1,229);
 IPAddress gateway(192,168,1,1);
 IPAddress subnet(255,255,255,0);
 
 void setup() {
  Serial.begin(115200);
  //BUG? with delay below, fails to connect to specified IP (except perhaps on first use)
  //delay(500); //problem happens with delays of 150 or more
  WiFi.config(staticIP, gateway, subnet);
  delay(100);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
 
  Serial.println();
  Serial.print("Connecting");
  while(WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("WiFi Connected, IP:");
  Serial.println(WiFi.localIP());
 }
 
 void loop() {
 }
```

Regards

lrmoreno007, 2018-02-28:
I don't have problems.

closed 2018-03-01 by devyte

hasenradball, 2018-08-21:
Hello I think I have understood the Problem, see description below:

1. The ESP has an autoreconnect function with does a reconnect with an previous established WLAN connection.
2. With inserting an delay(500) the autoreconnect function is able to etablish the connection wih previous settings.
3. Therefore the WiFi configuration after the delay() is redundant and not possible!
4. If you want that the WiFi Configuration works fine, you should set a WIFI_OFF before.

hasenradball, 2019-01-15:
I prefer to set at the first line within the setup:

WiFi.setAutoConnect(false);

The you will also have no problems anymore

nbenm, 2019-01-15:
Hi,

Which solution would be the best, power consuming talking:
1 - always turn off wifi
2 - turn off wifi only when we are not in "Power on" mode
3 - set AutoConnect to false

Thanks

hasenradball, 2019-01-15:
When you know you nerd wifi then boot with wifi.
Else boot without wifi.

d-a-v, 2019-01-15:
About power saving we now have `preinit()`
https://github.com/esp8266/Arduino/blob/8a64a1236f447009bd53fec4863aff9bf5428b59/libraries/ESP8266WiFi/examples/EarlyDisableWiFi/EarlyDisableWiFi.ino#L18

BTW, any good battery-powered example is welcome.

hasenradball, 2019-01-15:
I have one good example

hasenradball, 2019-01-15:
Take this adapter:
https://www.mikrocontroller-elektronik.de/esp12e-tutorial-einstieg-mit-dem-esp8266-modul/

and use a Voltage controller up to 6 V or up to 13 V.
Then you will be able to use a Photo akku.

nbenm, 2019-01-15:
@d-a-v I'm not sure this is a good example for you, but it's good enough for me.
I use this app with an esp8266 powered by a very small battery.
Please remove all "oled" instructions, they are power consuming. They are here only for test purpose.
Comments are in french, it was a private app.
It reads temp from a DHT22.
Serial is used only for debugging, it is disabled otherwise.
I've just uploaded source [here](https://github.com/nbenm/meteo_locale/upload/master)
I'll add schematic later.

happytm, 2019-12-15:
The lowest power ESP device I have tried is with following code:

https://github.com/happytm/BatteryNode

It is work in progress but I have tested it for about a month and I am hoping to run it for ever without ever changing coin cell battery I use (LIR2450). I use 53x30 mm micro solar panel to charge it using TP4056 charge controller. ESP8266 is bear bone ESP12 chip with BME280. Very inexpensive and most power efficient setup. The most power saving come from code as it requires very little uptime.
 
Thanks