esp8266/Arduino #3494: No Poison after the block Error in Smartconfig

romuye123, 2017-08-04:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.1.0-rc2

### Description
Downloaded the latest Smartconfig SDK on iOS/android. The smartconfig works perfectly fine for the first few times. During the testing i was repeatedly trying the smartconfig even if it is connected to WIFI and the ESP got a crash with poison block error mentioned below after 4th or 5th attempt. It happens randomly. Is it not advisable to repeatedly use smartconfig even if the device is connected to WIFI ? or is there any workaround for this ?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:	 SERIAL
Reset Method:		ck

### Sketch

```cpp
void setup()
{
Serial.begin(115200);
startSmartConfig();
}
void startSmartConfig(){

    WiFi.mode(WIFI_STA);
    WiFi.beginSmartConfig();
    delay(200);
  
    //Wait for SmartConfig packet from mobile
    Serial.println("Waiting for SmartConfig.");
    while (!WiFi.smartConfigDone()) {
      delay(500);
      Serial.print(".");
    }
  
    Serial.println("");
    Serial.println("SmartConfig received.");
  
    //Wait for WiFi to connect to AP
    int counter = 0;
    Serial.println("Waiting for WiFi");
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
      counter++;
      if(counter == 60){
        break;  
      } 
    }
    counter = 0;

    if(WiFi.status() != WL_CONNECTED)
    {
      Serial.println("WiFi not Connected.");      
    }
    else
    {
      Serial.println("WiFi Connected.");
      Serial.print("IP Address: ");
      Serial.println(WiFi.localIP());
    }
}
void loop(){}
```

### Debug Messages

```
Smart config activated...
Waiting for SmartConfig...................there is no poison after the block. Expected poison address: 0x3fff6e91, actual data: 0xa5 0xa5 0xa5 0x0
block start: 3fff6ab4
```

romuye123, 2017-08-04:
@me-no-dev need your help on this.

tablatronix, 2017-09-28:
I got this a few times, I also get wdt resets using 2.4.0

```
Waiting for SmartConfig.
scandone
scandone
20:b3:99:aa:3e:49: 583
20:b3:99:a9:f5:49: 583
20:b3:99:aa:6e:39: 639
20:b3:99:aa:69:89: 639
20:b3:99:aa:6e:39: 585

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
system_get_rst_info() reset reason: REASON_WDT_RST
```

JimmyTai, 2017-10-01:
I got this error message too. Could anyone help us?

Khizer-Jan, 2017-10-20:
this error is probably caused due to corrupt flash memory. In order to completely format flash use following code.
https://github.com/kentaylor/EraseEsp8266Flash/blob/master/EraseFlash.ino

Hope it will solve your problem.

romuye123, 2017-12-06:
@Khizer-Jan it did not solved the problem.
Still the same.

Khizer-Jan, 2017-12-07:
Well! it did for me. If you find any other solution ,kindly tell me about that.

earlephilhower, 2019-01-31:
Smartconfig is delivered by Espressif as a binary blob from the SDK.  There's nothing in our current code base which produces the error message quoted, so I would have to guess it's coming from inside that blob.  Nothing for us to do here.

closed 2019-01-31 by earlephilhower

yuanminglove, 2023-04-16:
我也遇到了这个问题，相同的代码使用linux可行，windows11不行
编译和上传环境都是使用 vscode+platformIO    太神奇了

tart...
No poison after block at: 0x3fff183c, actual data: 0xde 
【2023-52-16 14:52:09 946】0xff 0xa5 0xa5
User exception (panic/abort/assert)----------
【2023-52-16 14:52:10 012】----- CUT HERE FOR EXCEPTION DECODER ---------------

Abort cal