esp8266/Arduino #3495: Cannot program my ESP-12 module

anandhuskumar, 2017-08-04:
I am a beginner. I have used the Arduino 1.8.3 IDE for programming my ESP-12 module. I made the connection as shown below:
VCC     --------->   3.3V
TXD0   --------->   Rx of FTDI
RXD0   --------->   Tx of FTDI
GPIO0  --------->   pulled up through 1K resistor
GPIO2  --------->   pulled up through 10K
GPIO15   --------->   pulled down through 10K
RST       --------->   pulled up through 10K
EN    --------->   pulled up through 10K
Two push buttons are used to pull down RST and GPIO0.
I programmed the basic led blinking program (BUILTINLED) and is working fine. The pin that is blinking now is GPIO1(TXD0). It is working fine. But I cannot reprogram the module. 
If I connect the TXD0 to the Rx of serial to usb converter the module itself is not working. ie, if I press the RST button without connecting the TXD0 the module is getting restet (on board blue led blinks), the voltage level at TXD0 is 3.3V. But when I press the RST button after connecting the TXD0, the module is not responding and the voltage level at TXD0 pin drops to about 1.2V.

Please help and thanks in advance.

tttapa, 2017-08-04:
This issues section is meant for bugs, problems and questions regarding the ESP8266 Arduino core, not for troubleshooting your own project. 
To get help, you could try the Arduino and ESP8266 fora.
This [Beginner's Guide to the ESP8266](https://tttapa.github.io/ESP8266/Chap01%20-%20ESP8266.html) contains pretty much everything you need to know.

Pieter

devyte, 2017-09-05:
@anandhuskumar your first three words hint at your needing assistance, and this is not the right place for it. @tapa is correct, closing.

closed 2017-09-05 by devyte