esp8266/Arduino #3496: "delay(10)" after Serial.begin in examples

tttapa, 2017-08-04:
Many examples of the ESP8266WiFi library have this in the setup:
```cpp
Serial.begin(115200);
delay(10);
```
What is the purpose of the delay? At first glance, everything seems to work perfectly fine without it.
[GitHub search for "delay(10)"](https://github.com/esp8266/Arduino/search?p=2&q=++%22delay%2810%29%3B%22&type=&utf8=%E2%9C%93)

Thanks,
Pieter

5chufti, 2017-08-04:
this is leftover from Arduino Leonardo
`void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  // prints title with ending line break
  Serial.println("ASCII Table ~ Character Map");
}
`

tablatronix, 2017-11-04:
I usually use a 1 second delay so my serial program can reconnect to the serial port or else i miss the boot sequence and first print

5chufti, 2017-11-05:
a delay in the setup() won't give you time to see the boot message ...

tablatronix, 2017-11-05:
sorry I meant my setup boot sequence, not esp

bryankettle, 2018-10-16:
It's just left over code. It is not really needed nowadays. 
It is used to give time for the esp to initialize the serial

earlephilhower, 2019-01-16:
Fixed in #5290

closed 2019-01-16 by earlephilhower