esp8266/Arduino #3498: IGMP - Timer bug in igmp_start_timer()

forkineye, 2017-08-05:
### Issue
igmp_start_timer() sometimes assigns excessive times to group->timer, causing IGMP memberships to expire.  The problem is related to the usage of ```r_rand()``` via the ```LWIP_RAND()``` macro in the following function: https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/src/core/ipv4/igmp.c#L703.  ```group->timer``` should never exceed 100.  Debug output in the function produces the following:
```
os_printf("igmp_start_timer: %8X / %u / %u \n", &group->group_address, group->timer, max_time);
igmp_start_timer: 3FFF55B4 / 65515 / 100 
igmp_start_timer: 3FFF559C / 65 / 100 
igmp_start_timer: 3FFF5584 / 65493 / 100 
igmp_start_timer: 3FFF556C / 65503 / 100 
igmp_start_timer: 3FFF5554 / 43 / 100 
igmp_start_timer: 3FFF5054 / 65531 / 100 
```
### Fix
Assigning the output of ```LWIP_RAND()``` to a ```uint32_t``` variable and using that in place for the modulus fixes the issue.

Also, using RANDOM_REG32 in place of r_rand() works as well:
```
#define LWIP_RAND() *((volatile uint32_t *)(0x3FF20E44))  // RANDOM_REG32
```

Not sure why its breaking as is, but the above two fixes work for me.

forkineye, 2017-08-05:
Using os_random() instead of r_randr() is the best solution I've found.  PR submitted https://github.com/esp8266/Arduino/pull/3499

forkineye, 2017-08-07:
Thanks for merging the PR!

closed 2017-08-07 by forkineye

gjt211, 2017-08-07:
Is it possible for someone to provide some symptoms of the fault this fixed if possible?
Thanks.

forkineye, 2017-08-07:
The symptom was that devices would stop receiving multicast traffic after a while if associated with an AP that had IGMP snooping enabled.  I noticed it while testing a 720 pixel matrix being driven via multicast in my ESPixelStick project - https://github.com/forkineye/ESPixelStick.  I tested the fix for 48hrs streaming E1.31 data at 50ms refresh rates.