esp8266/Arduino #3500: sending file using wemos d1 mini

Bushii7, 2017-08-07:
i want to send files from one Wemos D1 mini to other D1 mini
how can i do that?

devyte, 2017-08-07:
This is an issue tracker for the core libs, not a support forum. Please ask at stackoverflow or esp8266.com and close this.

capedra, 2017-08-27:
@Bushii7 

FTP, WebSocket, HTTP, TCP, UDP, MQTT, CoAP, ESP-NOW, SoftwareSerial, I2C, SPI, HardwareSerial, UART0, UART1, UART2. Did I forget anything?

If you're going to connect them using UART (wisest choice), I suggest you to choose pins D4<->RX, RX<->D4. So, use Serial1 to send data on pin D4 of each device and Serial.read to read what's coming from the sender.  Also, do not forget to stop using LED_BUILTIN because LED_BUILTIN=D4 on Wemos D1 mini.

You also have the option to use pins D7<->D8, which is UART2, but it's not a good option because you have to call Serial.swap(). After you call that, though, you lose the serial monitor. So, you have to do Serial.swap() again to monitor the output of the serial port (in other words, to enable RX<->TX again).

closed 2017-12-28 by igrr