esp8266/Arduino #3504: Files cannot be opened for writing after some time

davidk88, 2017-08-08:
#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3.0-rc2

### Description

I'm writing some data into files every 10s. After some time files cannot be opened for writing ("w") anymore (it can be opened for reading). Problem persists until I upload filesystem again.

I also tried with lastest git version, but result is the same.

Thanks in advance

### Settings in IDE

Module:  			ESP-12E
Flash Size: 		1MB/4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL

devyte, 2017-08-08:
Please add a minimalist sketch to reproduce the issue.

davidk88, 2017-08-10:
I ran simple sketch which writes to a file every second and that is fine.
In my app where I have problems, it looks like that there is a problem with (low?) memory and then filesystem gets corrupted. This is Exception I get:

`Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 41 results
0x4010011d: cont_ret at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/cont.S line 118
0x4010068c: _umm_realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1584
:  (inlined by) realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4010020c: umm_assimilate_up at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1180
0x4010068c: _umm_realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1584
:  (inlined by) realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x40217434: ieee80211_recv_action_vendor_spec at ?? line ?
0x4021184f: ieee80211_send_setup at ?? line ?
0x40219954: wpa_auth_sm_event at ?? line ?
0x40213404: ieee80211_rfid_locp_recv_open at ?? line ?
0x40214d43: sta_input at ?? line ?
0x40211b48: ieee80211_send_nulldata at ?? line ?
0x402129e9: ieee80211_send_proberesp at ?? line ?
0x4010068c: _umm_realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1584
:  (inlined by) realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x40219028: wpa_auth_sm_event at ?? line ?
0x4020738e: SPIFFSImpl::open(char const*, fs::OpenMode, fs::AccessMode) at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/spiffs_api.cpp line 35
0x40214b04: sta_input at ?? line ?
0x402190d7: wpa_auth_sm_event at ?? line ?
0x40219379: wpa_auth_sm_event at ?? line ?
0x40212e48: ieee80211_freedom_output at ?? line ?
0x402193dc: wpa_auth_sm_event at ?? line ?
0x40213208: pwrsave_flushq at ?? line ?
0x4010068c: _umm_realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1584
:  (inlined by) realloc at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4010020c: umm_assimilate_up at \AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1180
0x40213251: pwrsave_flushq at ?? line ?
0x40231b59: ram_set_txbb_atten at ?? line ?
0x40215638: chm_return_home_channel at ?? line ?
0x4021568b: chm_set_current_channel at ?? line ?
0x40230000: chip_v6_initialize_bb at ?? line ?
0x4021ca57: rijndaelEncrypt at ?? line ?
0x4010570e: spi_flash_issi_enable_QIO_mode at ?? line ?
0x402318d4: cal_rf_ana_gain at ?? line ?
0x402318fa: cal_rf_ana_gain at ?? line ?
0x4023151f: set_rfanagain_dc_reg at ?? line ?
`

devyte, 2017-08-10:
You shouldn't be using 2.3.0-rc2, it's ancient. You should use 2.3.0
stable, or better yet: latest git.

Be aware of:
Max stack size for the ESP is 4KB (or 5, I can never get the straight of
that). Using the stack is very good for reducing heap fragmentation, but if
you go over the stack size limit you'll have memory corruption. If you have
a huge object on the stack, try creating it dynamically with new
(preferably into a smart pointer).
Don't declare huge strings as just const char *. Strings like that are
stored in RAM. Instead, (ab)use F and PROGMEM. Especially don't init
arduino String objects like so:
String blah = "some huge ass string that never ends zomg like those used
for static html content which is a Bad Idea on the ESP so don't do it";
If you are serving static html, store it in a file on the SPIFFS and let
the webserver stream it. Any dynamic content can be served as tiny jsons or
some other key=value pair format or w/e, and applied to the html doc via js
on the browser side. A js is another static file on the SPIFFS.
Beware your power supply, trust nothing. When in doubt, go big and heavy
for testing. I used a 5V 20A DC/DC converter, powered from a 200W AC/DC
converter. This proved to me that some DC power supplies that claim 1A or
2A like those used for cell phones are in fact crap, and not enough for the
ESP. That includes some iphone power adapter clones.
You code that executes in the loop() can't be of arbitrary duration. If you
take too long, the wifi stack will wind up and the buffers will overrun.
You can remedy the situation somewhat byof seeding with yield() calls, but
that helps only in cases of wiffi traffic not addressed to the ESP. Keep
your code small, seed itnwith ith yield calls, break it up into a state
machine to execute in small pieces if nothing else else works.

Please close this issue, adjust your design, instrument your mem usage
everywhere, and open a new one if you are sure there is a specific issue
that can be addressed.

closed 2017-08-14 by davidk88