esp8266/Arduino #3506: ESP8266 UDP transmission interruptions

pcc276, 2017-08-09:
### Basic Infos

#### Hardware
Hardware:			NodeMCU 1.0 ESP-12E 80MHz

### Description

Problem description
 I'm trying to transfer data from ESP8266(NodeMCU ESP12E) to my laptop. I'm coding ESP using Arduino IDE. I am able to receive correct packets (each containing 400 bytes) on a visual basic console program. However, to see whether the transmission is continuous or not, I have toggled an output pin whenever UDP.endpacket() returns 1. I observed waveforms on DSO. I found that UDP.endpacket() does NOT return 1 many times and the there is no transmission from ESP for a significant amount of time (milliseconds). Refer attached waveforms.Why does UDP.endpacket() return 0 for such long time? Why is UDP transmission getting interrupted? How can I solve this? Please feel free to ask for more details if necessary.
-- | --

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E module), 80MHz
Flash Size: 		4MB
CPU Frequency:       80Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

const char* ssid = "Connectify-me33";
const char* password = "donfour276"; 

WiFiUDP Udp;
unsigned int localUdpPort = 65000; 

unsigned int dest_port = 11000;

unsigned char clientBuf[400];
int clientBuf_len=0;

#define test_pin 5    

int return_udp_fcn = 0;

void setup()
{

  WiFi.begin(ssid, password);
  delay(5000);

  Udp.begin(localUdpPort);

  
  pinMode(test_pin, OUTPUT);
  digitalWrite(test_pin, HIGH); 
  
  pinMode(LED_BUILTIN, OUTPUT);     
  digitalWrite(LED_BUILTIN, LOW); 
  delay(1000);
  digitalWrite(LED_BUILTIN, HIGH); 
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW); 
  delay(1000);
  digitalWrite(LED_BUILTIN, HIGH); 
  delay(1000);

  
}

void loop()
{
        digitalWrite(test_pin, HIGH); 

        //filling up buffer
        for(clientBuf_len = 0; clientBuf_len<=399; clientBuf_len++)
        {
            clientBuf[clientBuf_len] = 0x7E;
        }

        //UDP transmit
        Udp.beginPacket("172.22.0.1", dest_port); 
        Udp.write((const uint8_t *)clientBuf, 400);
        return_udp_fcn = 0;
        return_udp_fcn = Udp.endPacket();
        if(return_udp_fcn == 1){
          digitalWrite(test_pin, LOW); 
        }
        delayMicroseconds(100);
        
}

![image](https://user-images.githubusercontent.com/30869832/29127592-2adf9bb4-7d3f-11e7-8f62-45e9dd0a3171.png)

RudyFiero, 2017-09-03:
I am using Arduino 1.8.4 and 2.4.0-rc1
I tried your code and I did not see what your capture shows. I got this.
[image]

AJITH-TOMY, 2017-11-15:
i got the similar issue, it looks more when connect my android phone in the same network.

d-a-v, 2017-11-15:
You should activate WIFI debug level. You have it for the Generic board or for every board in current master.
But you should realize that you are asking for unregulated(udp) ~30Mbits/s (400bytes/100us) so it is expected that the output buffer lose output packets before beeing able to send them.

d-a-v, 2017-11-18:
You should also try lwip2 (in current master) because we just realized that ```udp.endPacket()``` now answers 0 when the packet cannot be sent.

AJITH-TOMY, 2017-11-28:
i found that this is my router issue. (no QoS in WiFi)
no packet are losing.
router take some time to do some other WiFi function or internal function that why there i get a delay.