esp8266/Arduino #3507: Random Crashes on NodeMCU 1.0 / ESP8266

benni-germany, 2017-08-09:
### Basic Infos
Node MCU 1.0 / ESP8266

ESP-12E crashes Randomly.

Code is from [https://github.com/nassir-malik/IOT-ESP8266-ESP12E-Alexa-Multiple-Devices](https://github.com/nassir-malik/IOT-ESP8266-ESP12E-Alexa-Multiple-Devices)

Error codes at the end in Debug Section.

Just ask if thats not enough info, but i dont know what i could add. (-:
 
Thanks in advance (-;
#### Hardware
Hardware:			Node MCU 1.0 / ESP8266 ESP-12E
Core Version:      	?

### Description

Node MCU 1.0 / ESP8266 ESP-12E crashes randomly

error see debug info
### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB (3MSPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		?
Flash Frequency:	?
Upload Using:		OTA / SERIAL?
Reset Method:		ck / nodemcu?

### Sketch

Foll folder with all sketch data: [wemos.zip](https://github.com/esp8266/Arduino/files/1212437/wemos.zip)

 

wemos.ini
```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

// prototypes
boolean connectWifi();

//on/off callbacks 
void Steckdose1On();
void Steckdose1Off();
void Steckdose2On();
void Steckdose2Off();
void Steckdose3On();
void Steckdose3Off();
void Relay4On();
void Relay4Off();
void LEDOn();
void LEDOff();

// Change this before you flash
//#######################################
const char* ssid = "myssid"; //enter your access point/wifi router name
const char* password = "mypassword"; //enter router password
IPAddress ip(192, 168, 1, 100);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
[... 467 lines cut ...]

Exception (9):
epc1=0x40202777 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4f480a0d depc=0x00000000

ctx: cont 
sp: 3fff0f80 end: 3fff1160 offset: 01a0

>>>stack>>>
3fff1120:  3fffdad0 00000000 3fff0124 3fff012c  
3fff1130:  3fffdad0 00000000 3fff0124 40203454  
3fff1140:  3fffdad0 00000000 3fff0124 4020797c  
3fff1150:  feefeffe feefeffe 3fff0140 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
⸮

```

devyte, 2017-08-10:
Just from looking at your code, there are several issues and
inconsistencies.
You create objects on the heap with new, which is ok to save stack, but
then you declare huge Strings on the stack.
The ESP has a stringent stack limit of 4KB (or was it 5KB, I can never
remember).
You have Strings constructed from const char *, which then just get
concatenated into other Strings. Just declare them as const char * to
reduce RAM. Better yet, make good use of F and PROGMEM.
Then, const char * strings are stored in RAM, not program memory (unless
you use F). Declaring so many huge strings just for static html content is
a Bad Idea. Instead, you should put static html on the SPIFFS and serve it
as a file (the webserver doesand streaming for this). Any dynamic content
should be served as a tiny json or other type of key-value pair format
which then gets added to your html with js on the browser side. The js is
another static file.

Overall, I think you have a stack size issue, or a heap usage issue, or
both. Please close this and adjust your design. If then you are sure you
are within the ESP limits, and are still facing reboot issues, please open
a new issue.

3fff1120:  3fffdad0 00000000 3fff0124 3fff012c
3fff1130:  3fffdad0 00000000 3fff0124 40203454
3fff1140:  3fffdad0 00000000 3fff0124 4020797c
3fff1150:  feefeffe feefeffe 3fff0140 40100718
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
⸮

benni-germany, 2017-08-10:
Ok,
thank you

benni-germany, 2017-08-10:
Sorry could you lpease give me some examples what i should change, because im really new in arduino programming and i have no plan what you mean with this...

And it resets also with the original code.....

benni-germany, 2017-08-10:
I think i fixed the error by using this [https://bitbucket.org/xoseperez/fauxmoesp](https://bitbucket.org/xoseperez/fauxmoesp), wich is also much more easy to use, and only one file.

closed 2017-08-10 by benni-germany