esp8266/Arduino #3508: ESP8266 WiFi autoreconnect but server instance is lost.

pbecchi, 2017-08-10:
----------------------------- Remove above -----------------------------

### Basic Info
Hardware:		NODEMCU 1.2
Core Version:      	2.3.0-rc1

### Description

ESP8266 autoreconnect but Server is stopped.

In a poor signal status ESP8266 try to reconnect every second but if connected again server function is not automatically restored (e.g. server.begin() need to be re executed). 
 Is this a normal behavior?
 Can this be controlled somehow? 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		 SERIAL?
Reset Method:		 nodemcu?

### Sketch
```
bool EtherCardW5100::WiFiconnect( const uint8_t* my_ip, const uint8_t* gw_ip, const uint8_t* dns_ip, const uint8_t* mask)
{
	//if (WiFi.status() != WL_CONNECTED)
	{

		DEBUG_PRINTLN("Wait WIFI...");
		delay(1000);
		WiFi.mode(WIFI_STA);
		delay(1000);
		WiFi.begin(SSID, PASSWORD);// , my_ip, dns_ip, gw_ip);
		byte netmask[4] = { 255,255,255,0 };
		 WiFi.config(my_ip, gw_ip, netmask);
		DEBUG_PRINT("\nConnecting to "); DEBUG_PRINTLN(SSID);
		uint8_t i = 0;
		while (WiFi.status() != WL_CONNECTED && i++ <=201) { yield(); delay(400); DEBUG_PRINT('.'); }
		if (i >= 201) {
			DEBUG_PRINT("Could not connect to"); DEBUG_PRINTLN(SSID);
			return false;
		}

		DEBUG_PRINT("Server started IP="); DEBUG_PRINTLN(WiFi.localIP());
		return true;
	}
}

void setup(){
...............
	while (n < nNetwork&&!result) {
		SSID =  CharFromString(found_SSID[n]);
		PASSWORD =  CharFromString(found_psw[n]);
		n++;
		result = WiFiconnect( my_ip, gw_ip,  dns_ip,  mask);
	}	
if (result) {
		String outstr = "Conn. ";
		outstr+=	WiFi.SSID();
		message(outstr);
		incoming_server.begin();`
}
}
.............
```

### Debug Messages

```
pm open,type:2 0
state: 5 -> 2 (2c0)
rm 0
pm close 7
reconnect
state: 2 -> 0 (0)
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with Vodafone-Out, channel 5
ip:192.168.1.24,mask:255.255.255.0,gw:192.168.1.1
state: 5 -> 2 (2c0)
rm 0
reconnect
state: 2 -> 0 (0)
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with Vodafone-Out, channel 5
ip:192.168.1.24,mask:255.255.255.0,gw:192.168.1.1
state: 5 -> 2 (2c0)
rm 0
reconnect
state: 2 -> 0 (0)
f r0, scandone
no Vodafone-Out found, reconnect after 1s
reconnect
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (4)
reconnect
f -180, scandone
no Vodafone-Out found, reconnect after 1s
reconnect
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f -180, scandone
no Vodafone-Out found, reconnect after 1s
reconnect
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with Vodafone-Out, channel 5
ip:192.168.1.24,mask:255.255.255.0,gw:192.168.1.1
state: 5 -> 2 (2c0)
rm 0
reconnect
state: 2 -> 0 (0)
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with Vodafone-Out, channel 5
ip:192.168.1.24,mask:255.255.255.0,gw:192.168.1.1
state: 5 -> 2 (2c0)
rm 0
reconnect
state: 2 -> 0 (0)
f r0, scandone
no Vodafone-Out found, reconnect after 1s
reconnect
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with Vodafone-Out, channel 5
ip:192.168.1.24,mask:255.255.255.0,gw:192.168.1.1
state: 5 -> 2 (6c0)
rm 0
reconnect
```

igrr, 2018-01-02:
This has to be addressed either by documenting the existing behavior, or by modifying server classes to subscribe to WiFi events.

devyte, 2018-01-03:
My understanding is that the current usage model expects the user to make use of the WiFi events to restart any servers (e.g.: mdns notifyap).
Therefore, I'll address by documenting the behavior.