esp8266/Arduino #3510: ESP8266HTTPServer Does not parse some characters

MrD2, 2017-08-10:
### Basic Infos
Trying to send a POST with JSON body, there are some characters that are "eaten" by the parser (i guess).
Mainly when calling the function httpserver.arg("plain") the result should be:

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx%8hxxxxxxxxxxxxxxxxxxxxxxxx

but instead returns

xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

#### Hardware
Hardware:			ESP-12
Core Version:      	
hardware\esp8266\2.3.0
xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch (I can't post the all the things...

void WEB_SERVER::HandleConfigPost(void){
  //Check if body is json(does not really check)
  if (!p_server->hasArg("plain")){ //Check if body received
    SEND_CLIENT_ERROR_AND_RETURN;
  }

  WEB_SERVER_PRINTLN(p_server->arg("plain"));  <---- This is debug where i see that it "ate" the chars

//At this point accessing p_server->arg("plain") errors reboot because statement above i guess
  remote_debug->setEndpoint("/HandleConfigPost");  <----- Also debug but remote terminal print
  remote_debug->send(String(p_server->arg("plain"))); <----- Also debug but remote terminal print
  //Here we have body. Let's get it
  //Get body
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(p_server->arg("plain"));
  if (!(root.success() && root.containsKey("name") && root.containsKey("password"))) {
    SEND_CLIENT_ERROR_AND_RETURN;
  }
  //SAVE changes to file and reboot
  if(!wifi.configure_file(root["name"], root["password"])){
    SEND_CLIENT_ERROR_AND_RETURN;
  }
  //p_server->sendContent("HTTP/1.1 301 OK\r\nLocation: /\r\nCache-Control: no-cache\r\n\r\n");
  //TODO: REDIRECT HERE
  p_server->send(200);
  HandleReboot();
}

devyte, 2017-08-10:
There is already a PR that fixes some encoding/decoding issues. Have you
tried that?

MrD2, 2017-08-11:
> There is already a PR that fixes some encoding/decoding issues. Have you
tried that?

I cant see it.... can you post a link?

closed 2017-09-07 by MrD2