esp8266/Arduino #3512: Problem with esptool and arduino bin files

jonathanc56, 2017-08-12:
### Basic Infos

#### Hardware
Hardware:			?ESP-07?

### Description

I'm currently trying to create flash a .bin file created using the Arduino IDE with the esptool.

My test sketch is simple, basically a serial init and a write every second. I create the bin file using "Export Compiled Binary". 

Arduino settings:

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

So far so good.

I set up esptool.py and follow the instructions at https://github.com/espressif/esptool.

I then use the following line to try to write the .bin file to flash.

esptool.py --port COM3 write_flash 0x1000 testcode.bin

The console displays the following and all seems well:

esptool.py v2.0.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 228304 bytes to 165547...
Wrote 228304 bytes (165547 compressed) at 0x00001000 in 41.2 seconds (effective 44.3 kbit/s)...
Hash of data verified.

However, when I manually reset the device and check the serial monitor I get the following output (at 74880 baud)

~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010
~ld
v00001010

ad infinitum.

I tried other codes and get the same result. I tried manually setting up the flash mode and flash size detect and nada.

Could anyone help me figuring out why it isn't working for me? Any help on this issue would be appreciated.

suculent, 2017-08-12:
I have always written flash from 0x0000. What's so special here?

Odesláno z iPhonu

jonathanc56, 2017-08-15:
@suculent Yes, I found that the problem was that I started writing at 0x0001. It works fine when I start at 0x0000. 

I was following the example code on the esptool Github repository page (https://github.com/espressif/esptool)

Thanks for your help

closed 2017-08-15 by jonathanc56

suculent, 2017-08-15:
I've seen the docs yesterday and it appears 0x1000 would be correct if you had ESP32. 

You're welcome. 

Odesláno z iPhonu