esp8266/Arduino #3517: WifiClientSecure seems to not disconnect

odelot, 2017-08-14:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	update_sdk_2.1.0 branch (but I got with 2.3.0 as well)

### Description

After I remove the WLAN cable from the router, it continues writing with success and the connected method keeps returning true  (in the `client->state() == ESTABLISHED` case)

 after a while, the write method cannot allocate more memory and crashes. I am using the WifiClientSecure with ArduinoWebSockets

### Stack

```
Decoding 30 results
0x401004d8: malloc at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x4020632c: ClientContext::_write_from_source(DataSource*) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 615
0x40206e00: ClientContext::write(unsigned char const*, unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 615
:  (inlined by) __ax_port_write at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 644
0x402239bc: send_raw_packet at ssl/tls1.c line 2005
:  (inlined by) send_packet at ssl/tls1.c line 1267
0x40224524: ssl_read at ssl/tls1.c line 2005
0x40212256: optimistic_yield at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x40223a48: ssl_write at ssl/tls1.c line 2005
0x40209abe: WebSockets::sendFrame(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool, bool, bool) at C:\Users\fabio\Documents\Arduino\libraries\arduinoWebSockets\src/WebSockets.cpp line 97
0x40206880: WiFiClientSecure::write(unsigned char const*, unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 615
0x40209bce: WebSockets::sendFrame(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool, bool, bool) at C:\Users\fabio\Documents\Arduino\libraries\arduinoWebSockets\src/WebSockets.cpp line 241
0x40101f90: wDev_ProcessFiq at ?? line ?
0x4020a76c: WebSocketsClient::sendBIN(unsigned char*, unsigned int, bool) at C:\Users\fabio\Documents\Arduino\libraries\arduinoWebSockets\src/WebSocketsClient.cpp line 263
0x4010053d: _umm_realloc at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x402121ac: esp_yield at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x4020a78c: WebSocketsClient::sendBIN(unsigned char const*, unsigned int) at C:\Users\fabio\Documents\Arduino\libraries\arduinoWebSockets\src/WebSocketsClient.cpp line 263
0x4020b298: AWSWebSocketClient::write(unsigned char const*, unsigned int) at C:\Users\fabio\Documents\Arduino\libraries\aws-mqtt-websockets/AWSWebSocketClient.cpp line 476
0x40203521: MQTT::Client ::sendPacket(int, Countdown&) at C:\Users\fabio\Documents\Arduino\libraries\MQTTClient/MQTTClient.h line 328
0x40203a04: MQTT::Client ::publish(int, Countdown&, MQTT::QoS) at C:\Users\fabio\Documents\Arduino\libraries\MQTTClient/MQTTClient.h line 799
0x4010068c: free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40204cdc: MQTT::Client ::publish(char const*, void*, unsigned int, unsigned short&, MQTT::QoS, bool) at C:\Users\fabio\Documents\Arduino\libraries\MQTTClient/MQTTClient.h line 878
:  (inlined by) MQTT::Client ::publish(char const*, void*, unsigned int, MQTT::QoS, bool) at C:\Users\fabio\Documents\Arduino\libraries\MQTTClient/MQTTClient.h line 888
:  (inlined by) MQTT::Client ::publish(char const*, MQTT::Message&) at C:\Users\fabio\Documents\Arduino\libraries\MQTTClient/MQTTClient.h line 895
:  (inlined by) sendmessage(char const*) at C:\Users\fabio\Documents\Arduino\stayintouchinc8/stayintouchinc8.ino line 487
0x4010020c: _umm_free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40204db8: movementStatusMessage(bool) at C:\Users\fabio\Documents\Arduino\stayintouchinc8/stayintouchinc8.ino line 524
0x40212904: AWSWebSocketClient::connected() at C:\Users\fabio\Documents\Arduino\libraries\aws-mqtt-websockets/AWSWebSocketClient.cpp line 476
0x40204e20: iotNightLamp() at C:\Users\fabio\Documents\Arduino\stayintouchinc8/stayintouchinc8.ino line 714
0x40204eff: loop at C:\Users\fabio\Documents\Arduino\stayintouchinc8/stayintouchinc8.ino line 822
0x4020346c: setup at C:\Users\fabio\Documents\Arduino\stayintouchinc8/stayintouchinc8.ino line 781
0x402121f8: loop_wrapper at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x40100714: cont_norm at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109
```

cristovao-trevisan, 2017-10-19:
Probably the same as in #3537. There is also a fix that is not fully tested.

d-a-v, 2017-12-13:
Probably related to #3957 since memory is full and realloc() is involved.

earlephilhower, 2018-09-11:
Can't reproduce under git head.  `client.write()` seems to fail cleanly when there is no WiFi connection.  The crash shown in the report looks like it's a standard OOM under axTLS with a large user app.

Closing as can't reproduce.  If you can make a MCVE, please file a new one so we can look at it.

My simple tester (connecting to an openssl s_server):
````
#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin("....", "....");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  WiFiClientSecure client;
  client.connect("192.168.1.8", 4433);
  client.write((const uint8_t*)"HELLO\n", 6);
  WiFi.begin("aaa", "bbb");
  int c=0;
  while (1) {
    client.write((const uint8_t*)"HELLO\n", 6);  
    Serial.printf("%d\n", c++);
    delay(1);
  }
  client.stop();
}

void loop() {
  // put your main code here, to run repeatedly:

}
````

closed 2018-09-11 by earlephilhower