esp8266/Arduino #3519: Can I use Arduino to develop ESP-07S?

dereckgo, 2017-08-15:
### Basic Infos

#### Hardware
Hardware:			ESP-07S
Core Version:      	2.3.0

### Description
 I  developed ESP-12S using arduino for collection data from STC51. but when i replace ESP-12S with ESP-07S ，using the same software and hardware , i find program can not run. I write print code  in  setup(), but it turn out print nothing， so i doubt the code in setup() and loop() is not running

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			Dio
Flash Frequency:	40Mhz
Upload Using:		none
Reset Method:		ck

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

adams13x13, 2017-08-22:
Hi dereckgo,
according to your sketch, it works correctly - does nothing. It would be much easier to help if you wuold provide your code and some details about connection.
Possible problem: if you have serial interface and use DTR/RTS to automatically enter programming state, your serial terminal program can set these lines to a wrong state, like RESET or PROGRAMMING. Disconnect the interface or check levels of RESET and GPIO 0.
If you try to activate a LED, it can be connected to another port.

devyte, 2017-09-03:
Without looking at any details whatsoever, I suggest looking at the pinouts. The pins may not be the same, in which case you may need to adjust your sketch.
When you change board, it is recommended to figure out the pins first. Look at a pinout diagram, and actually test which pins can be set to output, input, whatever, without hanging the module, and which are the real serial port pins.

underwoodblog, 2017-09-03:
I've replaced an ESP-12E with an ESP-07S on an Nodemcu-Board without any problems. The 07S may have a smaller flash size.

devyte, 2017-09-05:
Per st. google, the 07S has 4MB flash, same as the 12E/F/S.
If you replaced it on a Nodemcu board and it worked, it hints at something wrong with your connections, and not in the core.

closed 2017-12-27 by igrr