esp8266/Arduino #3521: Can't compile using fs.h in a class

enny1170, 2017-08-15:
### Basic Infos

#### Hardware
Hardware:		wemos d1 mini
Core Version:      	1.20300.1

### Description

I'm try to use FS.h in my class, see Code below. every time im include them i got fallowing Compiler error.

In file included from C:\Users\e.ziehe\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:25:0,
from lib\config\src/ConfigBase.h:13,
from lib\config\src\ConfigBase.cpp:8:
c:\users\e.ziehe\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\hashtable.h: In constructor 'std::_H
ashtable<_Key, _Value, _Alloc, _ExtractKey, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits>::_Hashtable(_InputIterator, _InputIter
ator, std::_Hashtable<_Key, _Value, _Alloc, _ExtractKey, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits>::size_type, const _H1&, c
onst _H2&, const _Hash&, const _Equal&, const _ExtractKey&, const allocator_type&)':
C:\Users\e.ziehe\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id
 before '(' token
#define max(a,b) ((a)>(b)?(a):(b))

I can use fs.h directly in my .ino file. This will compiled successfully. 
I searched around but can't find a solution or helpfull Suggestion. So i'm guess this is caused by my includes or order of includes. I'm not an expert in c++. Please help to fix this.

### Settings in IDE

Module:  			wemos d1 mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

///////////////////////////////////////////////////////////
//  ConfigBase.h
//  Implementation of the Class ConfigBase
//  Created on:      10-Aug-2017 23:42:46
//  Original author: e.ziehe
///////////////////////////////////////////////////////////

#ifndef _CONFIGBASE_H_
#define _CONFIGBASE_H_

// #undef min
// #undef max
#include <FS.h>
#include <ArduinoJson.h> //https://github.com/bblanchon/ArduinoJson
//#include <Arduino.h>     //https://github.com/arduino/Arduino
#include <ESP8266WiFi.h>
//#include <Hash.h>
#include <unordered_map>

/**
 * Minimum Implementation for JSON Configuration
 */
class ConfigBase
{

public:
	ConfigBase();
	virtual ~ConfigBase();
	char* Key;
	/**
	 * This is a Hashtable of Params as String, you can Add by using Params_Add an
	 * take the Value by Params_Get. Also Iteration is possible by fallowing Code
	 * for(auto (and)itr : ParamsList)
	 * {
	 *   Printf("key: %s, value: %s",itr.first,itr.second);
	 * }
	 */
	std::unordered_map<const char*,const char*> ParamsList;
	//std::list<String> ParamsList;

	virtual char* GetKey();
	virtual String GetRawConfigFile(char* filename);
	virtual void SetKey(char* newVal);
	/**
	 * Build a Json String from Configuration Object, has to be overwritten in Your
	 * implementation. Have in your mind if isRoot so we have to build a valid
	 * Document, Is not we have to build as String with Key
	 */
	virtual String BuildConfigString(bool isRoot = false);
	/**
	 * Writes the _rawConfiguration Json String to SPIFS File
	 */
	virtual void WriteRawConfigFile(char* filename);

	/** 
	 * Parse the Content of a Json String and setup ParamsList
	 */
	bool ParseConfigString(String jsonString);
	bool ParseConfigFile(char* filename);
	//void Params_Add(String key, String newValue);
	//void Params_Add(char * key,char * newValue);
	void Params_Add(const char *key, const char * newValue);
	//void Params_Add(const char *key, char * newValue);
	//char* Params_Get(String key);
	//char* Params_Get(char * key);
	const char* Params_Get(const char * key);
	
	/**
	 * This Method can be used to Set DefaultConfig shoul be overwritten in your Implementation
	 */
	virtual void SetDefaults(bool write=false);
private:
    String _rawConfiguration;
};
#endif 
```

### Debug Messages

```
messages here
```

philbowles, 2017-08-15:
I believe this is due to the way the stdlib redefines the min and max macros - I've had the same problem myself. There are several solutions around if you srat searching around that as the source of the problem

suculent, 2017-08-15:
It's rather "FS.h" instead of <FS.h>

M. 

Odesláno z iPhonu

enny1170, 2017-08-16:
I try this, but the same error.

enny1170, 2017-08-16:
What i found on Internet sounds like this. But in can’f find any description how to work around this. Maybe you can Point me in the right direction.

suculent, 2017-08-16:
Are you including <Arduino.h> before that in your library?

enny1170, 2017-08-17:
this are the includes in my library class. As i understand the arduino.h will included automatically into the main .ino file during compilation.

#include <Arduino.h>
#include <FS.h>
#include <ArduinoJson.h> //https://github.com/bblanchon/ArduinoJson
#include <ESP8266WiFi.h>
#include <Hash.h>
#include <unordered_map>

This causes also the Error:
In file included from C:\Users\e.ziehe\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:25:0,
from lib\config\src/ConfigBase.h:13,
from lib\config\src\ConfigBase.cpp:8:
c:\users\e.ziehe\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\hashtable.h: In constructor 'std::_Hashtable<_Key, _Value, _Alloc, _ExtractKey, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits>::_Hashtable(_InputIterator, _InputIterator, std::_Hashtable<_K
ey, _Value, _Alloc, _ExtractKey, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits>::size_type, const _H1&, const _H2&, const _Hash&, const _Equal&, const _ExtractKey&, const allocator_type&)':
C:\Users\e.ziehe\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
#define max(a,b) ((a)>(b)?(a):(b))

devyte, 2017-09-17:
@enny1170 I believe there is a similar bug in Arduino.h with #include <vector>. Try this workround in your configbase.h:
```
#ifdef max
#undef max
#endif 
#ifdef min
#undef min
#endif 
#include <unordered_map>
```

devyte, 2017-10-22:
Closing as duplicate of #398 (same issue)

closed 2017-10-22 by devyte