esp8266/Arduino #3522: <time.h> library is missing strptime()

jxmot, 2017-08-15:
### Basic Info

#### Hardware
Hardware:			NodeMCU 1.0 (ESP-12E)
Core Version:      	2.3.0

### Description

I need to use the strptime() function to create an epoch-like value that represents midnight of the current day. However it looks like that strptime() is not present in the library because it appears that the build fails during linking.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E)
Flash Size: 		4MB/1MB SPIFFS
CPU Frequency:	80Mhz

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <time.h>

unsigned long midnight;
struct tm tm;

void setup()
{
    
    if (strptime("17-07-14 23:59:59", "%y-%m-%d %H:%M:%S", &tm) == NULL) midnight = 0;
    else {
        time_t t = mktime(&tm);
        if(t == -1) midnight = 0;
        else midnight = (unsigned long)t;
    }
    Serial.printf("midnight = %ld", midnight);
}

void loop()
{
    yield();
}
```

### Debug Messages

```
Archiving built core (caching) in: C:\Users\jim\AppData\Local\Temp\arduino_cache_718300\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M1M_7498f438d4d25dc3a43396406e659217.a
sketch\strptime_test.ino.cpp.o:(.text.setup+0x18): undefined reference to `strptime'

sketch\strptime_test.ino.cpp.o: In function `setup':

E:\Workspaces\IoT\ESP8266\arducam_K0061_ESP8266_KIT\tests\time\strptime_test/strptime_test.ino:10: undefined reference to `strptime'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

JMishou, 2017-12-24:
I have the same problem, however; I can most definitely see that the function is there in the the core libs.

https://github.com/esp8266/Arduino/search?utf8=%E2%9C%93&q=strptime&type=

Did you ever figure this out?

jxmot, 2017-12-24:
No I have not. I also tried implementing a replacement using sscanf() but that's also missing from the 8266 core. I think that one or the other might be corrected in 2.4.0, but I can'r say for sure.

devyte, 2017-12-24:
@jxmot your issue says 2.3.0 core, which is rather old. Please retest with latest git.

jxmot, 2017-12-24:
I can't right now. I'm using the release 2..3.0 and unfortunately too busy to uninstall it and install a release candidate. Or is it possible for the two versions to coexist in the IDE?

devyte, 2017-12-24:
Only one version can be installed at a time.
Latest git != release candidate.
Uninstalling and later reinstalling 2.3.0 is just a few clicks from the Arduino IDE. Installing latest git requires a few commands from a console, but it's documented. Uninstalling just means deleting the created install dir.

JMishou, 2017-12-24:
I'll install the git library later this evening and report back.  I've also been using the latest ide version.

devyte, 2017-12-25:
Don't forget to restart the Arduino IDE after installing the core via git

JMishou, 2017-12-25:
I installed the current git version and strptime compiles!  Thanks! 

jxmot it was really simple...  Literally a 4-5 minute process.  There are more options for the boards (at least for the D1 mini) it took me a minute to notice that you have to enable the serial port.   

devyte... any idea why it doesn't work in 2.3.0?  I can see the function in the time library... either way I don't see why this issue should remain open.

devyte, 2017-12-25:
2.3.0 release was quite some time ago,  and a whole lot has been fixed since.
Closing due to already fixed.

closed 2017-12-25 by devyte