esp8266/Arduino #3524: Can't upload code on NodeMcu.

thetweny, 2017-08-16:
Few weeks back I have tested my new NodeMcu with a few sketches. And when I tried to upload sketch now it says: warning: 
espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
Even if i try to flash bin file it doesn´t show up mac adress as it used to. 
I have tried to use it as programmer with my arduino mini pro to see if ch340g module is dead and it worked. I have also tried all possible combinations of settings of baud rate, flash freq. , cpu freq. ,etc...
I have tried to put it in flash mode and connecting GPIO0 to GND and than flash, but it won´t upload anything too.

coelner, 2017-08-30:
try the native esptool from terminal and reset the device while esptool is waiting. Use this command: `esptool.py chip_id`
the system should be responding. 
If the system answers you should hold the button "flash" down, reset the nodemcu, release the hold button and upload then the sketch. Do not replug the usb, the old udev generated entry get probably lost and the arduino enviroment fails about that.

devyte, 2017-09-05:
The error: espcomm_open failed message is an indicator of a user error. Either you have something wrongly connected, wrongly configured, or you are doing something wrong.
Example: for Nodemcu boards, you have to press reset while holding the flash button, then let go of the flash button. Only then can you flash.
This is not the place to request assistance for such things. Closing.

closed 2017-09-05 by devyte