esp8266/Arduino #3525: If POST JSON.stringify($('form').serialize()) the first value is omitted.

nardev, 2017-08-16:
Maybe it's unusual to submit for in that way, but just to report  that if you  use AJAX and submit form in one value as stringify and serialized value in it, the first value will be omitted..

For example if you put:

```
...
data: JSON.stringify($('form#mqtt').serialize()),
...
```
of Ajax code with POST

Than on device if you print:
```
  Serial.println(server->arg("hostname"));
  Serial.println(server->arg("port"));
  Serial.println(server->arg("username"));
  Serial.println(server->arg("password"));
```
The hostname will be omitted.. or any other first value... probably because there is no &sign, which i assume is used as delimiter or  so...

NdK73, 2017-08-19:
Maybe my understanding on what you're doing is is wrong (if so, please post some more details), but shouldn't you JSON-decode server->arg("data") ?

nardev, 2017-08-19:
Actually, this is jquery js thing.. yes server gets data ok but doesn't recognize the first one.. anyway... just notice..

devyte, 2017-08-30:
I don't understand the issue. Please put a minimal sketch that shows what you are trying to do.
As far as I know, the ESP8266Webserver does not decode json. It received the json data as a string, and puts it in arg "plain", so you would get it on the server side like this:
DynamicJsonBuffer buffer;
JsonObject & json = buffer.parseObject(server.arg("plain"));

In addition, the ESP8266Webserver can't handle simultanous requests. In other words, if it is handling e.g.: a POST, and a GET comes in, it may not behave correctly. I was using it exclusively until recently, but I was forced to switch to the AsyncWebserver, because modern browsers are incapable of serializing HTTP requests.

nardev, 2017-08-30:
It's ok, you can delete this thread...

vicnevicne, 2017-08-30:
We don't "delete this thread". Instead, could you please "close the issue" on the GitHub page ?

closed 2017-08-30 by nardev