esp8266/Arduino #3526: Sensor Programming over I2C Bus

javiersenas, 2017-08-16:
Hello colleagues,

I am using a sensor connected to the ESP8266 via I2C whose FW (hex file) needs to be preflashed before the main code runs.  Does any of you have an idea if I can use the ESP8266 to flash the FW (the hex file) of this sensor? I mean, is there any tool which routes a file from UART to I2C? Any suggestion is welcome.

Thanks in advance!

suculent, 2017-08-17:
It would be worth mentioning which sensor. 

Depending on the firmware size you could store data on ESP,  turn the sensor to flashmode and send fw byte-by-byte. Depends on sensor documentation. 

Odesláno z iPhonu

javiersenas, 2017-08-17:
Sorry, I forgot it. This is a capacitive sensor Cypress' CY8MBR3008. Could
you please give more details about tools and procedures?

devyte, 2017-09-05:
@javiersenas your project sounds interesting, but this is not the place to discuss it. This is an issue tracker for tracking issues in the core libs. Please ask for assistance/discussion at esp8266.com or stackoverflow.

closed 2017-09-05 by devyte