esp8266/Arduino #3527: WPA2-enterprise + PEAP

manzanazo, 2017-08-16:
How sells firmware to esp8266 with WPA2-enterprise + PEAP
Regards

devyte, 2017-08-16:
The description of this issue makes absolutely no sense. Either improve it and add details, or close it.
If your intent is to use wpa2 enterprise, look through the PRs and open issues, there are some examples available.

manzanazo, 2017-08-17:
Hi everybody,
I bought this module esp8266 wifi thinking to be able to advance in my project. We use Cisco Aironet 3700 Series Access Points, having a secure WPA2-enterprise + PEAP. But the command that we execute to integrate us to the Wi-Fi network, only asks ssid and password. And I have been looking for some update for this module and I found this website. And I have seen that a lot of you have already been working on the issue, and since I have not seen any trade that offers such an update I thought that if someone from here already made changes and worked, I could buy.
I bought this module esp8266 wifi thinking to be able to advance in my project. We use Cisco Aironet 3700 Series Access Points, having a secure WPA2-enterprise + PEAP. But the command that we execute to integrate to the Wi-Fi network, only asks for ssid and password. And I've been looking for some updates for this module and I came across this website. And I have seen that a lot of you have already been working on the issue, and since I have not seen any trade that offers such an update I thought that if someone from here already made changes and worked, I could buy.
Since I wrote this post, I studied the characteristics of the AP and it could be that there we could find the solution but could not confirm it right now.

Devyte, thanks for your attention.

devyte, 2017-08-17:
The wpa2 enterprise topic has been asked several times, and there is a PR
somewhere with a reportedly working example, although I'm not sure about
PEAP. I suggest searching for that and giving it a try.
To try it, you have to use the latest git version of the core libs, and
pull the PR on top. If you don't know how to do that, there is an
explanation in readthedocs on how to use latest git, and research git for
pulling a PR, there are tutorials everywhere.

ninjabe86, 2017-08-17:
this issue has a few workarounds in the following thread
https://github.com/esp8266/Arduino/issues/1032

devyte, 2017-09-05:
@manzanazo any progress?

manzanazo, 2017-09-05:
no yet.

closed 2017-09-05 by manzanazo

averri, 2018-05-03:
Why is this issue closed??? The problem has not been addressed, there are uncountable people asking for a solution to connect to WPA2-enterprise + PEAP. This is a very basic and mandatory requirement!

d-a-v, 2018-05-03:
Because it's not an esp8266/arduino issue.
It has to do with espressif's firmware on which we do'nt have control.

https://github.com/espressif/ESP8266_NONOS_SDK/search?q=enterprise&type=Issues

which among other things relinks back here (as mentionned above, did you follow the link?)
[#1032(comment)](https://github.com/esp8266/Arduino/issues/1032#issuecomment-321812646)

manzanazo, 2018-05-03:
Hello. The solution for my project was to use the esp8266 module without using access validation for each divice, an access point and a computer for each line, and also a tcp serial bridge.
Regards