esp8266/Arduino #3528: Advanced Captive Portal Not Connecting in some PC's

santhosh8438, 2017-08-17:
Hi, I have used Advanced Captive portal for my code
In this the configuration page is not connecting(192.168.4.1) in some PC's,
I have disabled Anti-Virus in some PC's it's working well, But in some PC's it's not connecting 
I have checked with Windows 7,8 and 10 all have this issues!!!

This is my Code : 

#include <TimeLib.h> 
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266HTTPClient.h>
#include "user_interface.h"
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <ESP8266mDNS.h>
#include <EEPROM.h>
#define ind_led 2
#define AC 4
#define L1 5
#define L2 12
#define L3 13
#define FAN 14

/* Set these to your desired softAP credentials. They are not configurable at runtime */
const char *softAP_ssid = "my_ESP";
const char *softAP_password = "123456789";

/* hostname for mDNS. Should work at least on windows. Try http://esp8266.local */
const char *myHostname = "esp8266";

/* Don't set this wifi credentials. They are configurated at runtime and stored on EEPROM */
char ssid[32] = "";
char password[32] = "";

// DNS server
const byte DNS_PORT = 53;
DNSServer dnsServer;

// Web server
ESP8266WebServer server(80);

/* Soft AP network parameters */
IPAddress apIP(192, 168, 4, 1);
IPAddress netMsk(255, 255, 255, 0);

/** Should I connect to WLAN asap? */
boolean connect;

/** Last time I tried to connect to WLAN */
long lastConnectTry = 0;

/** Current WLAN status */
int status = WL_IDLE_STATUS;

int dimming = 128;  // Dimming level (0-128)  0 = ON, 128 = OFF

int check_bit = 0;
String check_byte;
String req_data;
String MAC_ID = "CHM";
const char* host = "webapp.scoto.in";
String req_url ="/demo/homeauto/api/response/deviceget/deviceID/"+MAC_ID;
os_timer_t myTimer;
int httpPort = 80;
//const char ssid[] = "CIPL";              //  your network SSID (name)
//const char pass[] = "14062016";          // your network password

IPAddress timeServer(132, 163, 4, 101);  // time-a.timefreq.bldrdoc.gov

[... 482 lines cut ...]
  server.sendHeader("Pragma", "no-cache");
  server.sendHeader("Expires", "-1");
  server.send ( 404, "text/plain", message );
}

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		ESP8266 12E
Core Version:      	esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_921600,FlashSize_4M3M

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 12E Module
Flash Size: 		4MB
CPU Frequency:		80Mhz

skx, 2017-09-06:
Please file your bug against the repository for the captive portal software you're using - this repository is for the esp8266 project itself, not random libraries that run upon it :)

treii28, 2018-04-01:
I can't speak for the original poster, but I'm not so sure it's specific to a single project though. I have tried literally every captive portal example I have found on the internet, and I haven't been able to get one of them to work. There's an active thread right now over on the WiFiManager github and it refers back to the core code as a possible source of the problem.

themindfactory, 2018-04-02:
I have implemented it this way in the past and when I connect to Android it
typically works but not always, on IOS its hit or miss and on Linux/Windows
its never.

Out of all that code only

*dnsServer.setErrorReplyCode(DNSReplyCode::NoError);dnsServer.start(DNS_PORT,
"*", apIP);
server.on("/generate_204", handleRoot); //Android captive portal. Maybe not
needed. Might be handled by notFound handler.
server.on("/fwlink", handleRoot); //Microsoft captive portal. Maybe not
needed. Might be handled by notFound handler.

are the important parts...

nicogon, 2018-04-20:
Did some test with a raspberry and I got it working by 
-Setting the DNS to resolve all queries  to 172.217.28.206 (public ip of http://clients3.google.com/)
-Then have a tcp rule to forward all port 80 traffic to my host 
-> captive portal working in all platforms. 
The problem with esp libraries is that I haven't been able to forward outgoing 80 tcp traffic to esp host. 
Is there any way  of doing this?

mcspr, 2022-07-12:
Should be covered with DNS server update and updated captive portal examples

closed 2022-07-12 by mcspr