esp8266/Arduino #3529: SPI Extended Mode

JohnGENZ, 2017-08-18:
On page 10 of "ESP8266 Arduino Core Documentation Release 2.4.0" there is a comment "There’s an extended mode where you can swap the normal pins to the SPI0 hardware pins. This is enabled by calling SPI.pins(6, 7, 8, 0) before the call to SPI.begin()."

When I try and compile this I get an error that reads "'class SPIClass' has no member named 'pins'" and when I look at the source I cannot see any implementation of pins.

Does anyone have any examples of using the extended mode or suggestions on how to use it.  I have the SPI working on the standard pins but want to use the extended pins if possible

tttapa, 2017-08-21:
It was added about four months ago:
[https://github.com/esp8266/Arduino/commit/af0f5ed956b85e41da647d51d1799db469d6e697](https://github.com/esp8266/Arduino/commit/af0f5ed956b85e41da647d51d1799db469d6e697)
Are you sure that you are using version 2.4.0. Check SPI.h on your computer to see if the pins method exists.

JohnGENZ, 2017-08-22:
This seems to have been the problem - not sure what went wrong but all fixed now.  Very many thanks.

closed 2017-08-22 by JohnGENZ