esp8266/Arduino #3530: WiFiClientSecure.h seems to collide with timers?

KrisZane, 2017-08-18:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.3.0?

### Description
For some odd reason I seem to not be able to use WiFiClientSecure inside an os_timer, whenever I do it inside the loop, using the counter from the os_timer it runs just fine, but move it into the timer itself and it burps up an exception. I can't seem to find any issue regarding this, and all I can do is guess that it is some sort of odd scope error?

### Settings in IDE

Module:  			?NodeMCU 0.9?
Flash Size: 		?4MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
void scanStatus() {
  WiFiClientSecure client;
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  String url = "/raw/jazz";
  Serial.println(host + url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");
         
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
  }
  
  String line = client.readStringUntil('\n'); // Skip the encoding mark
  line = client.readStringUntil('\n');\
  previousLed = activeLed;
  activeLed = line.toInt();
}

void cycleLed() {
  brightness = brightness + fade;
  String colour[] = {"R", "G", "Y", "A"};

  if(activeLed == 0) {
    analogWrite(r, brightness);
    analogWrite(g, 0);
  } else if(activeLed == 1) {
    analogWrite(r, 0);
    analogWrite(g, brightness);
  } else if(activeLed == 2) {
    analogWrite(r, brightness);
    analogWrite(g, brightness);
  } else if(activeLed == 3) {
    analogWrite(r, 0);
    analogWrite(g, 0);
  }

  Serial.print(colour[activeLed]);

  if (brightness <= 0 || brightness >= 255) {
    fade = -fade;
  }
}

void timerCallback(void *pArg) {
  if(tenMiliTimer % 3 == 0) {
    cycleLed();
  }
  if(tenMiliTimer == 1000) {
    scanStatus();
    Serial.println("Scanned");
  }
  
  if(tenMiliTimer >= 1000) {
    tenMiliTimer = 0;
  }
  tenMiliTimer += 1;
  
  //cycleLed();
}

void setup() {
  // Setup button and LED's, plus reset
  pinMode(r, OUTPUT);
  pinMode(g, OUTPUT);
  pinMode(b, OUTPUT);
  digitalWrite(r, 0);
  digitalWrite(g, 0);
  digitalWrite(b, 0);
  pinMode(button, INPUT);

  // Set up wifi
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  os_timer_setfn(&myTimer, timerCallback, NULL);
  os_timer_arm(&myTimer, 10, true);
  Serial.println("WiFi:" + WiFi.localIP().toString());
}

void loop() {
  buttonCheck();
}
```

### Debug Messages

```
Exception (29):
epc1=0x40231004 epc2=0x00000000 epc3=0x00000000 excvaddr=0x15d01468 depc=0x00000000

ctx: sys 
sp: 3ffffd70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff10:  3ffec5b6 3fff1664 3fff17b4 40230fa4  
3fffff20:  3fff12bc 3fff1fa4 3fff1fa4 3ffeef50  
3fffff30:  00000000 3fff17b4 0000001c 3fff1fa4  
3fffff40:  3ffec5a8 00000000 3fff17b4 402303ad  
3fffff50:  8080800a 00000090 00000000 0000001e  
3fffff60:  00000002 0000001a 40206b7f 3ffede30  
3fffff70:  3ffec580 3fffdcc0 3ffea460 3ffea460  
3fffff80:  40206af2 3ffede30 00000000 3fff200c  
3fffff90:  3fffdc80 00000000 3fff17b4 4021abbb  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```

JamesGKent, 2017-08-18:
you cannot have long duration functions inside a timer callbacks as that stops the wifi functions from doing their work, you need to set a flag in your callback and process it in your loop

KrisZane, 2017-08-20:
@JamesGKent That actually makes a lot of sense, thanks. :)

closed 2017-12-27 by igrr