esp8266/Arduino #3534: reset after a few minutes of running

ejri, 2017-08-20:
### Basic Infos

Hi, I'm wondering if I can get some help on some issues I'm having. 
The sketch is a counter using two ultrasonic sensors. When one of the sensors is triggered followed by another trigger at the other sensor (within a few seconds), the counter either goes up or down. The new count number is uploaded to ubidots. 

The ESP runs for about 10minutes before it resets. 

The serial monitor sometimes shows that it's a reset issue (rst cause:2, boot mode:(1,6)) or a watchdog issue (rst cause:4, boot mode:(1,6)).  

I tried creating a watchdog in the code like in one of SensorsIot's videos (#33 Internet of Things with ESP8266 #5: Watchdogs, Timers & Stability on youtube
https://www.youtube.com/watch?v=D_7ciW_TCac).

This makes it run for a longer time before it resets. 

I have tried unplugging the LCD (the LCD uses GPIO0 and GPIO2 pins, because it's connected via i2c), adding pull up resistors, capacitors, switching the ESP unit.

The ultrasonic sensors are connected to GPIO 12, 13, 14, and 15. They are also connected to an external 5.0V source. 

Unplugging the LCD makes the ESP run for a few more minutes, I don't know why...  

### Settings in IDE
Arduino 1.8.0
Module:  			NodeMCU 1.0 ES12-E Module  
Flash Size: 		4MB
CPU Frequency:		80Mhz

### Sketch
```

#include <Wire.h> 
#include <LiquidCrystal_I2C.h>
#include <ESP8266WiFi.h>
#include <SimpleTimer.h>

#include <Ticker.h>
Ticker secondTick;
volatile int watchdogCount=0;

void ISRwatchdog(){
  watchdogCount++;
  if (watchdogCount==5){
    ESP.reset();
    }
  }

// the ultrasound stuff
#define trigPin1 D7
#define echoPin1 D8

#define trigPin2 D5
#define echoPin2 D6

#define led D6
#define led2 D5
#define led3 9
int hit1 = 0, hit2 = 0;
int numEntries = 0, numExits = 0;int i=1; int k=1;
unsigned long startTime_after_1;
unsigned long startTime_after_2;

// LCD stuff
char array1[]="Hello!!               "; //the string to print on the LCD
char array2[]="             "; //the string to print on the LCD
int lcd_tim = 500; //the value of delay time
// initialize the library with the numbers of the interface pins
LiquidCrystal_I2C lcd(0x27,16,2); // set the LCD address to 0x27 for a 16 chars and 2 line display

// ultrasound stuff

// Define and initialize constants and variables that we'll use later in the code
[... 789 lines cut ...]

after I unplug the LCD,

```
Panic /Users/ibrahim/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:131 loop_task

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffff80:  4021ac9a 3fffdab0 00000000 3ffee990  
3fffff90:  00000000 3fffdad0 3ffee988 402046dd  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

Decoding 5 results
0x4021ac9a: ets_timer_handler_isr at ?? line ?
0x402046dd: loop_task at /Users/ibrahim/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
```

devyte, 2017-08-22:
Your loop body looks huge. While I do see a few yields in there, I somehow
doubt they're enough.
I suggest timing the loop body to figure out how long you're hogging the
cpu and keeping it away from the wifi stack. There's a limit to that, and
behavior depends on wifi traffic.

On Aug 20, 2017 5:28 PM, "Ibrahim" <notifications@github.com> wrote:

> Basic Infos
>
> Hi, I'm wondering if I can get some help on some issues I'm having.
> The sketch is a counter using two ultrasonic sensors. When one of the
> sensors is triggered followed by another [...]

ejri, 2017-08-23:
Thank you for the reply. 
Yeah, I don't know if the loop can shortened. 

I was thinking of hooking up the sensors to a nano, and connecting the nano as an i2c slave to the ESP, but I think this setup requires less power...

devyte, 2017-08-24:
Yes it can, you just have to implement it as a state machine,
e.g.: setup(): init state to 0
void loop()
{
switch(state)
{
case 0:
 handleSensor0();
 ++state;
 beak;
case 1:
  handleSensor1();
  ++state;
  break;
etc.

with the above, the execution time of one loop() will be much smaller,
which gives the wifi stack more breathing room.

Also, use good coding practices, as in break up your code into smaller
functions, then implement the core logic or sequence by calling those
functions. You can then add/remove the function calls one by one and figure
out which is causing the problem. And add heap checks in between, and debug
msgs.

As it is, your code is one huge blob that is too hard to understand, let
alone debug.

Check that the pins of your board are really correct. Check your power
supply, I use a 5V 20A (as in twenty amps) power supply for development, to
be sure any resets or comm hangs are not caused by voltage dips (I've had
many of those).

Having said that, this is not the right forum to discuss these things. This
is an issue tracker. Please close this, adjust your design, and if you
track a problem to a piece of code in the core libs, open a new issue with
a minimal code example.

ejri, 2017-08-24:
Thank you for the detailed response!! I will try your suggestions out.

closed 2017-08-24 by ejri

suculent, 2017-08-25:
What happens if the state is stored in a 2-byte value and is incremented above its max value (65535 for unsigned and 32768) for signed int?

Crash, very likely. Or reset back to 0/-32767. 

Odesláno z iPhonu

devyte, 2017-08-25:
If you have several thousand states, your design is far too wrong.

You have a finite small numbernof states, each of which represents a small
piece of a long sequence of code. The last state is the last to execute,
theb instead of ++state you do state=0, and your sequence starts over.

suculent, 2017-08-26:
You're right. I did mean counter++ of course. 

I was just trying to point that certain values may overflow unexpectedly (byte, long, etc. )

Odesláno z iPhonu