esp8266/Arduino #3535: esp 8266 wdt reset

wxlf366, 2017-08-20:
hi im using esp8266 and i got this error

`chksum 0x2d
csum 0x2d
v09f0c112
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
`
can help me. im new in arduino and this for my final year project.

this is my testing code

 `#include <AccelStepper.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

#define FULLSTEP 8

const byte moto1 = 7;
const byte moto2 = 8;
const byte moto3 = 9;
const byte moto4 = 10;

const byte moto5 = 3;
const byte moto6 = 4;
const byte moto7 = 5;
const byte moto8 = 6;

char ssid[] = "DVAZ40A";
char password[] = "dvaz40axs";  // your network key

int c = 0;

AccelStepper stepper1(FULLSTEP, moto1, moto3, moto2, moto4);
AccelStepper stepper2(FULLSTEP, moto5, moto7, moto6, moto8);

int steps1 = 0;
int steps2 = 0;

void setup() {
  Serial.begin(115200);
  
  stepper1.setMaxSpeed(3000.0);
  stepper1.move(1); 
  stepper1.setSpeed(1000);
  
  stepper2.setMaxSpeed(3000.0);
  stepper2.move(0);
  stepper2.setSpeed(1000);
  Serial.begin(115200);
 

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  // attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

}

void loop() {
  c = Serial.read();

  if (c == '1') {
    stepper1.move(4200);
    stepper1.setSpeed(1000);
  }
  else if (c == '2') {
    stepper1.move(-4200);
    stepper1.setSpeed(1000);
  }
  else if (c == '3'){
    stepper2.move(4200);
    stepper2.setSpeed(1000);
  }
  else if (c == '4'){
  stepper2.move(-4200);
  stepper2.setSpeed(1000);  
  }
  
  
  steps1 = stepper1.distanceToGo();
  stepper1.runSpeedToPosition();
  steps2=stepper2.distanceToGo();
  stepper2.runSpeedToPosition();
  
}

`

thanks in advance and sorry for my bad english

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1M(64K)
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:	 SERIAL
Reset Method:		ck

asetyde, 2017-09-05:
Please search in repository a useful guide to understand esp boot erro code.
For wdt you can find useful information looking for in issues search box.

devyte, 2017-09-05:
This looks like a lack of understanding of the ESP's use semantics. A run through the loop() function can't take more than a certain amount of time, or the wdt will trigger. This is on purpose, because the wifi stack must be serviced every so often.
Please ask at esp8266.com or stackoverflow for assistance and discussion, and also do some research about the esp8266, and the reasons behind the wdt.

closed 2017-09-05 by devyte

suculent, 2017-09-05:
I have looked into the wdt implementation inside Esp.cpp today:

1) hardware wdt reset will happen 8 seconds after using wdt_disable()
2) setting wdt_enable(65535) soon after disabling wdt to expand soft timeout to over one minute helps solving many network-related soft wdt timeout issues (e.g. requests that take more than 8 seconds to happen, decrypt and decode)

Hooe this helps to close. 

Odesláno z iPhonu