esp8266/Arduino #3538: I want to solve this problem "error: espcomm_open failed"

Eliyanow, 2017-08-21:
# I want to solve this problem 

`Arduino: 1.8.2 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 9600, 4M (3M SPIFFS)"

Archiving built core (caching) in: /var/folders/9w/78rv0fq92rq4123npr0b4y_00000gn/T/arduino_cache_801873/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_9600,FlashSize_4M3M_5eec79ddd5008eadfb7afd203997fcef.a
Sketch uses 585561 bytes (56%) of program storage space. Maximum is 1044464 bytes.
Global variables use 54844 bytes (66%) of dynamic memory, leaving 27076 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

JamesGKent, 2017-08-23:
this line:
error: espcomm_open failed
indicates that it couldn't open the port to the ESP, have you set the correct port in the menu in the arduino IDE?

tttapa, 2017-08-24:
Did you install the CH340G USB drivers?

Eliyanow, 2017-08-25:
yes I install the CH340G USB !! but the same problem

devyte, 2017-09-05:
This looks like a user error of one type or another. Either the drivers are missing, wrong port is selected, or something else that needs to be done is missing. 
AFAIK, flashing from MacOS is currently not broken, so this is not an issue with the core.
Please ask at esp8266.com or stackoverflow for assistance and discussion.

closed 2017-09-05 by devyte

adamaero, 2018-04-20:
I had the same problem. Then I tried a different cord. Use a **data acquisition cord**, not a power cord.

drmpf, 2018-04-20:
Could be that your ESP8266 is not in programming mode.
I recently used an ESP-01S Link V2.0  UART to USB board  that failed to program the attached ESP8266 because the Link V2.0 was not pulling GPIO0 low on reset.
Fine for looking at serial output and sending AT commands but not for programming via Arduino IDE

sheysson, 2018-04-20:
Ok

miguelalbuquerque96, 2018-05-30:
Nothing works i got i Wemos d1 r1 that was working just fine and sundundy i coudnt uploaded it with new programms, i  always get the same error
 warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
what can i do ?

ravisayal, 2019-01-06:
I was also facing this issue.
Refer the following note great information
https://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html

KusalThiwanka, 2019-05-05:
> Nothing works i got i Wemos d1 r1 that was working just fine and sundundy i coudnt uploaded it with new programms, i always get the same error
> warning: espcomm_sync failed
> error: espcomm_open failed
> error: espcomm_upload_mem failed
> error: espcomm_upload_mem failed
> what can i do ?

I also get this error. did you find any solution? Thank you

adamaero, 2019-05-05:
Use a data acquisition cord, not a power cord.

Slayer4477, 2020-02-18:
change tx pin to gpio ,,, i solve this problem with ota upload

void setup() {
  Serial.begin(9600);
  while (!Serial);
WiFi.hostname("xxxxx");      // DHCP Hostname (useful for finding device for static lease)
  ArduinoOTA.setHostname("xxxx");
  ArduinoOTA.setPort(8266);
  ArduinoOTA.setPassword("");
  ArduinoOTA.onStart([]() {

    /////////////////change pin to gpio////////
    //GPIO 1 (TX) swap the pin to a GPIO.
    pinMode(1, FUNCTION_3);
    //GPIO 3 (RX) swap the pin to a GPIO.
    pinMode(3, FUNCTION_3);
    ///////////////////////
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    ////////////////change pin to rx tx again//////////
    //GPIO 1 (TX) swap the pin to a TX.
    pinMode(1, FUNCTION_0);
    //GPIO 3 (RX) swap the pin to a RX.
    pinMode(3, FUNCTION_0);
    ///////////////////////
    Serial.println("\nEnd");
  });
  ArduinoOTA.begin();
  delay(1000);
}