esp8266/Arduino #3540: WIFI mostly not working

Drtuna, 2017-08-22:
Hi,

I have an NodeMCU Development Board. I try to connect the Board to my WIFI Network. However, the connection mostly doesn't work. Basically the code gets stuck in the connection loop (    while(WiFi.status() != WL_CONNECTED) {)

The only exception is if I flashed the NodeMCU firmware just before I flash the code from the Arduino IDE. After that the code connects immediately. Also if I do a reset it will reconnect. But only until I unplug the power. When I replug the Power It won't work anymore. Also if I flash my code a second time it will not work

### Basic Infos

#### Hardware
Hardware:			AZDelivery NodeMCU Lua Amica Module V2 ESP8266 ESP-12E
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB (3MB SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			n/a
Flash Frequency:	n/a
Upload Using:		 SERIAL (USB)
Reset Method:		n/a

### Sketch

```cpp

#include <Arduino.h>

void setup() {
...
    // Connect to WiFi network
    Serial.print("Connecting to ");
    Serial.println(ssid);
    IPAddress ip(192, 168, 2, 80); // where xx is the desired IP Address
    IPAddress gateway(192, 168, 2, 1); // set gateway to match your network
    Serial.print(F("Setting static ip to : "));
    Serial.println(ip);
    IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your
    WiFi.config(ip, gateway, subnet);
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
    Serial.println("WiFi connected");
}

void loop() {

}

bill-orange, 2017-08-24:
I am having what may be the same problem with a ESP12-F using substantially the same sketch.  I can only connect after a power-off of around 2 minutes.  It is as if the module is not transmitting.

Drtuna, 2017-08-24:
(By googling) I figured out the problem and a work-around. The problem seems to be that the RF module is not properly initialised when the device is powered on or awakes from reset.

Sadly there seems to be no manual mode to switch on the RF module.

However, I found a workaround. First I made the connection to enable deep sleep, for that, we need to tie the `RST` pin to `D0`/`GPIO 16` on the ESP8266. 

Then I added the following code to setup 

    extern "C" {
      #include "user_interface.h"
    }
    
    void setup(){
      if (resetInfo->reason != REASON_DEEP_SLEEP_AWAKE) {
    
          ESP.deepSleep(10, WAKE_RF_DEFAULT)
      }

Basically whenever the system comes into setup from something else than deep sleep, the system goes to deep sleep and when powering back on the RF module is enabled.

bill-orange, 2017-08-24:
Interesting. I found my problem to be a power supply issue. The 3.3 v supply had a 1000 uf filter cap. If I turned the unit off and back on WiFi would not connect. If I tuned the unit off and shorted the cap (there was a small spark, it stayed chargered) and tuned the unit back on WiFi connected flawlessly. The cap appears to be located on the regulated side (wrong) and does not have a drain resistor. Being lazy I simply removed the cap. The board is a T5 Blackboard.

ehlen000, 2018-02-14:
For me it was USB power too..... adding a seperate powersupply solved the problem

earlephilhower, 2019-01-31:
I'm not seeing anything here other than some hardware (power, as usual) issues.  Nothing for us to do here.

closed 2019-01-31 by earlephilhower

MatsBerggren, 2019-02-16:
Try to add   WiFi.setPhyMode(WIFI_PHY_MODE_11B);
This solved a similar problem for me, the problem seems to be issues with the router and 20/40 mhz switching