esp8266/Arduino #3542: Download sketch from device

diffstorm, 2017-08-23:
Hello,

I can upload new .bin file into esp8266 using esp.local/update web server (ESP8266HTTPUpdateServer).

What I need is to download the .bin file from esp8266, imagine there will be a URL like esp.local/download/sketch.bin, when I enter this url on my browser, I will download the .bin file of the esp8266. Is that possible?
I'll be glad if you help me about it.

Regards.

tablatronix, 2017-08-24:
Do you want to actually read the flash to file or store a binary file in spiffs as storage?

The prior can be done with esptool

diffstorm, 2017-08-24:
I have 2 ESP boards, one connected to another. When I have updated first one's sketch, I want that second one automatically can read first one's sketch and update itself using ESPhttpUpdate.update("http://<first_esp_IP_address>/ota/sketch.bin");
Of course the will communicate first to decide about versions etc.. That's out of scope.

tttapa, 2017-08-24:
Why can't you just OTA them all at once?
The following script looks for all Arduino OTA devices on the network, prompts a BIN file and an OTA password, and then uploads the binary to all ESPs simultaneously:
```bash
echo "Scanning for Arduino OTA devices on the network ..."
ip_arr=($(((avahi-browse _arduino._tcp --resolve --parsable --terminate) 2>/dev/null | grep -F "=;") | cut -d\; -f8))
if [ ${#ip_arr[@]} == 0 ]
then
    echo "No devices found"
    exit
fi
echo "IP addresses found: "
printf '\t%s\n' "${ip_arr[@]}"

read -p "Path to binary: " path
read -p "Password: " password

read -p "Press [ENTER] to start OTA"
echo    "--------------------------"

for ip in ${ip_arr[@]}
do
    python2 /home/$USER/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/espota.py -i $ip -p 8266 --auth="$password" -f "$path" 2> /dev/null && echo -e "Success:\t$ip" || echo -e "Fail:   \t$ip" &
done
wait
```
Pieter

diffstorm, 2017-08-25:
Because they all are not in the same network. First is connected to wifi modem, second esp is connected to first one etc..
So first one should update the second esp or second esp should download .bin from first esp.

So how to share .bin over http?

suculent, 2017-08-25:
Webserver?

Odesláno z iPhonu

diffstorm, 2017-08-25:
@suculent How can share .bin (sketch) of the esp using web server as a file? That was my original question. How?

_Sent from my Samsung SM-N910C using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

tablatronix, 2017-08-25:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino

diffstorm, 2017-08-25:
@tablatronix Thank you, the streamFile method should do what I need. Only I do not know how to share the firmware of the device. So streamFile to retrieve the sketch (.bin) of ESP8266 directly without storing it in SPIFFS area.

Sorry about the misuse of this system, but indeed this became helpful for me, thanks for understanding. I'll take care about usage from now on.

I'll be glad if you answer my last question. Thanks.

tablatronix, 2017-08-25:
I think you are gonna have to use im not sure if you can read the flash from itself

tablatronix, 2017-08-25:
Wait maybe you use how ita backups

diffstorm, 2017-08-25:
@tablatronix Sorry, I didn't get your idea, what did you mean?

tablatronix, 2017-08-25:
Typo i meant ota.
Ota backups current flash then swaps the new, but i have no idea how, maybe look into that

suculent, 2017-08-26:
Use any http server example and replace index.html with the binary file, change content-type to application/binary. 
This will work with ESP8266httpUpdate as this needs no authentication and has no complex protocol like ArduinoOTA. 

Odesláno z iPhonu

diffstorm, 2017-08-26:
@suculent Thanks, I'll try this. First I need to find how to read the firmware of ESP itself.
Any info will be appreciated.
Thank you all!

suculent, 2017-08-26:
You can use esptool - read_flash instead of write_flash

diffstorm, 2017-08-30:
@suculent I need to do this programatically in the ESP itself. I'll take a look at esptool source code.

suculent, 2017-08-30:
I'm afraid ESPTool won’t help you then. Task would require a HTTP Server running on the ESP8266 master, that would be able to serve contents of ESP’s EEPROM instead of SPIFFS file. You can look into ESP8266httpUpdate to see how it WRITES to the flash and then find respective READ command. You need to stream it byte by byte.

The client side would be ESP8266httpUpdate library that is able to update firmware with plain HTTP GET request.

diffstorm, 2017-08-30:
I guess you mean the Updater.cpp file, I read it but it seems it's a bit hard to understand at first sight. I'll examine it deeply.
Following piece of code should do the actual write operation;
```
    eboot_command ebcmd;
    ebcmd.action = ACTION_COPY_RAW;
    ebcmd.args[0] = _startAddress;
    ebcmd.args[1] = 0x00000;
    ebcmd.args[2] = _size;
    eboot_command_write(&ebcmd);
```
Respectively there is a read function `int eboot_command_read(struct eboot_command* cmd);`
I'll try to find more information.
By the way, we can prepare an example for reading the .bin from ESP with this experience as well. So this topic would be helpful for others too.

diffstorm, 2017-08-31:
Hi guys, I was able to send binary file byte by byte simply using the following code;
On the computer side, I can download a file called bin and it's content is the same with bin array.
```
  server.on("/bin", []()
  {
    char bin[] = {0x55, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0, 0xAA};
    char i;
    
    for (i = 0; i < sizeof(bin); i++)
    {
      server.sendContent_P(&bin[i], 1);
    }
  });
```

devyte, 2017-09-01:
I don't see an issue with the core libs here, and I think that the
functionality that you need is already present. You just need to develop
your app.
Please close this issue, and continue your discussion on a forum like
stackoverflow or esp8266.com, if needed.

tablatronix, 2017-09-01:
Yeah Close it and comment here still makes no difference

diffstorm, 2017-09-01:
Thanks everyone, I see that the functionality I need does not present.
I'm closing it, thanks for your valuable comments.

closed 2017-09-01 by diffstorm