esp8266/Arduino #3543: Question: Http requests within isr

tonobo, 2017-08-23:
Is ist possible to use httpclient or wificlient or wifisecretclient within interrupt routines. I've discovered that i'm alway unable to open a connection within it. Within main loop all seems working

davisonja, 2017-08-23:
I'd be very surprised if it worked.
Aside from supporting code reentrancy, it feels like a tcp connection would
take far too long.

tonobo, 2017-08-23:
But the isr will be detached during runtime

tonobo, 2017-08-23:
I also tried to send a udp packet but this also won't work

tttapa, 2017-08-23:
In an ISR, interrupts are disabled. You need interrupts for the TCP/IP and WiFi stuff to work.

If your ISR code does more than setting volatile flags, you're doing something wrong, and you need to rethink the design of your program.

Pieter

closed 2017-08-24 by tonobo

suculent, 2017-08-25:
No that's not possible. You must store state in ISR only as it is not allowed to allocate any memory. You perform requests based on that state from main loop. 

Odesláno z iPhonu