esp8266/Arduino #3544: Can ESP-12E support MQTT with Client Certificates

vishalkothari, 2017-08-24:
Hello,

I have scanned multiple threads on this site and elsewhere. But I am not able to find concrete information about whether ESP-12E with axtls can support MQTT with client certificates.

I tried below code with firmware version 2.3.0 and latest github version.

```
const` char* mqtt_server = "test.mosquitto.org";
int mqtt_port = 8884;
WiFiClientSecure espClient;
PubSubClient client(mqtt_server, mqtt_port, callback, espClient); 

  File cert = SPIFFS.open("/client.crt.der", "r");
  if(!cert) {
    Serial.println("Couldn't load cert");
    return;  
  }
  Serial.println(cert.size());
  Serial.println(cert);
  if(espClient.loadCertificate(cert, cert.size())) {
    Serial.println("Loaded Cert");
  } else {
    Serial.println("Didn't load cert");
    return;
  }

  File key = SPIFFS.open("/client.key.der", "r");
  if(!key) {
    Serial.println("Couldn't load key");
    return;  
  }
  
  if(espClient.loadPrivateKey(key, key.size())) {
    Serial.println("Loaded Key");
  } else {
    Serial.println("Didn't load Key");
  }  
  
   espClient.connect(mqtt_server, mqtt_port);
   String clientId = "thing1";
   clientId += String(random(0xffff), HEX);
  
    if(client.connect(clientId.c_str())) {
      Serial.println("Connected");
      //client.subscribe("test");
      client.publish("abcd", "hello world ssl");
    } else {
      Serial.println("Not connected");
    }
```

With 2.3.0 release codebase, I got connect failed rc=-2
With latest github version, I got exception as below.
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 24 results
0x40212123: __memcpy_aux at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S line 92
0x4024c8b0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40225a7f: SHA512_Final at crypto/sha512.c line 207 (discriminator 3)
0x4021e318: x509_new at ssl/x509.c line 201
0x40100a22: ppEnqueueRxq at ?? line ?
0x40101f40: wDev_ProcessFiq at ?? line ?

Please note that I have tried certificates in both PEM and DER format.
Any clear answer or help in this regard will be very helpful.
Thanks in advance.

Best Regards.

copercini, 2017-08-24:
There are few examples: https://github.com/copercini/esp8266-aws_iot
I will improve the documentation soon

suculent, 2017-08-25:
ESP8266 is very slow to handle TLS/AES. You should use ESP32 for SSL. That's why you cannot find it - it's not worth trying. 

Odesláno z iPhonu

copercini, 2017-08-25:
I agree with @suculent 
For SSL/TLS ESP32 have a lot of new features, and much more heap to deal with it, so the things are more fast and stable

vishalkothari, 2017-08-25:
@suculent @copercini Thanks for your replies. You would advice ESP32 even for TLS 1.1 as well?

suculent, 2017-08-25:
> 1 @suculent @copercini Thanks for your replies. You would advice ESP32 even for TLS 1.1 as well?

TLS 1.1 is considered insecure. What you really want is TLS 1.2. 

https://www.esp32.com/viewtopic.php?t=137

Odesláno z iPhonu

vishalkothari, 2017-08-25:
Thanks and Is this the right link for ESP-32 board schematics? https://dl.espressif.com/dl/schematics/ESP32-Core-Board-V2_sch.pdf

copercini, 2017-08-25:
Yep, there are some more docs: http://esp-idf.readthedocs.io/en/latest/hw-reference/modules-and-boards.html

vishalkothari, 2017-08-25:
@copercini and @suculent Thank you so much!

closed 2017-08-25 by vishalkothari