esp8266/Arduino #3546: Send DHT data to HTML on ESP8266 webserver

kbdrox, 2017-08-24:
I have an HTML file with the formatting I want in a .h file.

This sends the HTML formatted just like I want it :

void handleRoot() {
  server.send(200, "text/html", index_html);
}

I want this variable inserted into the HTML when requested :

  temp = dht.readTemperature();
  tempString=String((int)temp);

so it will the temp reading from the DHT to my browser with the formatting I want.

This newb is all googled out.

Thanks in advance.

tttapa, 2017-08-25:
Don't change the HTML file, insert a (static) JavaScript that sends a second HTTP request to the ESP, gets the DHT data, and inserts it into the web page.
It's a good idea to store your static files in the SPIFFS instead of RAM (or progmem) char arrays.
[SPIFFS](https://tttapa.github.io/ESP8266/Chap11%20-%20SPIFFS.html)
[Temperature logger with static HTML files and dynamic web interface using JavaScript](https://tttapa.github.io/ESP8266/Chap16%20-%20Data%20Logging.html)

Pieter

devyte, 2017-08-25:
This is not the right place to ask such questions. This is an issue
tracker, meant to track bugs and feature requests in the core libs.

Please close this and ask in a forum like stackoverflow or esp8266.com.

kbdrox, 2017-08-25:
Sorry for the faux pas and thanks for the links. Much appreciated!

closed 2017-08-25 by kbdrox