esp8266/Arduino #3548: 2.4.0-rc1: ESP8266WebServer::sendContent() "crashes" when passed empty String()

arantius, 2017-08-25:
#### Hardware
Hardware:			ESP-12
Core Version:      	2.4.0-rc2

### Description

I have a pre-existing code base written on 2.3.0.  It has a setting which it stores as a `String` so that it can be passed to `ESP8266WebServer::sendContent()`.  The default value for this string is empty; I'm putting it in an HTML form for editing.

If the current value is empty string, and I run `webServer.sendContent(my_setting);` (in between other calls to send the HTML input around this value) the connection is terminated, no further content (despite more calls to `.sendContent()`) is sent. If it's non-empty, everything continues as expected.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

JamesGKent, 2017-09-05:
i suspect you'll find that the rest continues to be sent, but is not read/used by the webbrowser on the other end as an empty chunk is used to signal the end of a chunked transfer, so you are unintentionally sending and "end" signal...

arantius, 2017-09-07:
Mayhaps, but 2.3.0 did not chunk, so acted differently.  And the fact that 2.4.0 chunks is a private implementation detail.

This broke my existing code in a surprising way.  It might be worth not breaking existing users?

earlephilhower, 2020-01-01:
Refers to a specific >2yr old RC release and no update in 2 yrs.  Closing as nothing to do here.

closed 2020-01-01 by earlephilhower