esp8266/Arduino #3550: Client.connect, reading data via wifi

beaumr2, 2017-08-26:
Hardware:			ESPDUINO

Problem description:

Trying to get my ESPDUINO to read data from an OBDII->WIFI adapter in my car. I can get the client to connect and send/receive data when I run this code with the wifi in my house but it doesn't work on the OBDII wifi. I'm doing this to hopefully get inputs of the engine coolant temp, rpm, and throttle position. Any help greatly appreciated!!!!

Settings in IDE:

Module:  			ESPDUINO (ESP-13 Module)
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:		80Mhz
Upload Using:		wchusbserial.1420

Code: 

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <SPI.h>

//const char* ssid = "WiFi_OBDII"; // wifi network name
//const char* password = ""; // password 
const char* ssid = "FiOS-KEGEO"; // wifi network name
const char* password = "cost7624copy26mind"; // password 
byte ip[] = {192,168,1,1}; // find this on your wifi settings - it's the ip for the router. I change this when using the OBDII wifi
//char servername[]="google.com";
WiFiClient client;

void setup() {
Serial.begin(115200);

WiFi.persistent(false);
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
unsigned long startMillis = millis();
unsigned long currentMillis = millis();

Serial.print("Connecting to network...");
while (WiFi.status() != WL_CONNECTED) {
  delay(500);
  Serial.print(".");
  currentMillis = millis();
  if ((currentMillis - startMillis) > 30000) { 
    Serial.print("error"); //return error code for not connecting within 30 seconds
    break; 
  }
}
Serial.println(" Success!");
Serial.print("IP Address is: ");
Serial.println(WiFi.localIP()); //IP Address of your ESP8266
Serial.println(client.connect(ip, 80));
  if (client.connect(ip, 80)) {
    Serial.println("connected");
    //client.println("GET /search?q=dog HTTP/1.0");
      // client.println();
    return;
  }
}

void loop() {
  
 // while(client.available()) {
 for (int i = 0; i < 14; i++) {
    String line = client.readStringUntil('\n');
    Serial.println(line);
    if ( i == 13) {
      Serial.println("14 line breaks");
      delay(500);
      Serial.end();
    }
 }
    /*
    if (line == "Canis") {
      Serial.println("HELLO WORLD");
      delay(1000);
      Serial.end();
    } */
  /*
 if (client.available()) {
    char c = client.read();
    Serial.print(c);
  } */
  
client.flush(); 
}

tcpipchip, 2017-09-06:
OBD2 uses a diferent network! Your pc must to enter in that network!
Btw, does your obd2 wifi detects TCP timeout ? For example:
-Connect the esp8266 to wifi (TCP) (socket 35000 port)
-Power off the esp8266 or reset...
-Does your OBD2 wifi module detects that esp8266 was power off or was reset ? and then the obd2 WIFI closes socket ?

My obd2 wifi does not close the sockets and i must to reset it to accept new connections :(

beaumr2, 2017-09-06:
I'm able to connect to WiFi_OBDII with wifi.begin but not with client.connect

My OBDII wifi does not detect TCP timeout. Should I be using port 35000 instead of port 80 like I have in my code above? Also how would I tell if my OBD2 wifi knows if the ESP8266 was reset? I haven't been able to read any information from it at all

tcpipchip, 2017-09-06:
Late i will post my arduino example! I have changed the original project (that uses a obd2 uart module) and ported it to wifi.
I can connected my nodemcu to obd2 wifi sucessfully, read PID, show on display....but...if my nodemcu resets, the obd2 not close the sockets...then  i cant connect again until resets the obd2... :(

I have to find out a obd2 wifi that has time out for tcp connections...if no communication for a while, close the sockets :(

beaumr2, 2017-09-08:
Could you post your example please?

beaumr2, 2017-09-26:
You get any success?

tcpipchip, 2017-09-26:
Yes, got! Only looking for a obd wifi that has tcp timeout config

Enviada do meu iPhone

tcpipchip, 2017-09-27:
I lost the source where i ported from OBD2 SERIAL to OBD2 WIFI.

I am rebuilding now! If you want contact me, i am in tcpipchip@hotmail.com

Miguel

earlephilhower, 2019-07-25:
Looks like this issue was solved by @tcpipchip (seems to have been a clone OBD2 WiFi issue and not core related, in fact).  Closing.

closed 2019-07-25 by earlephilhower