esp8266/Arduino #3551: ESP8266 NodeMCU not detected by computer.

oemd001, 2017-08-27:
Hello I purchased a esp8266 NodeMCU and I connected the micro USB into the NodeMCU and connected via USB to my computer, which is running Windows 10 x64 bit. Nothing is showing up. However, the blue light on the esp8266 is flashing (occasionally) but no signal whatsoever from the computer.

Please help and respond?

Thanks!

killa198921, 2017-09-08:
Did you install the USB driver ?
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

NikolozCh, 2017-10-28:
Same problem, i have downloaded the drivers but still nothing :/ the blue lightning is flashing just

devyte, 2017-10-28:
Either your board is bad, or your drivers are bad. The ones with ftdi can suffer from fake chips, which then get bricked on purpose by the original ftdi driver.
I have several Nodemcu boards, and they flash fine.
This is an issue tracker for the core, and not the right place to ask for support. Please refer to a community forum.
Closing per #3655 .

closed 2017-10-28 by devyte

SACHINKUMARHACKER, 2018-03-26:
Same situation happened with me, I also purchased a nodemcu board of amica which is not showing on my windows platform.

pipi61, 2018-03-26:
check in device manager, if installed the usb driver, check serial port number,
and set this comp port number in arduino ide

sshocked, 2018-04-01:
I had same problem but I solved it. My board flashed only flashed once(it all about I think what my usb wire works) but, pc still doesnt recognize it. So i installed like 1 thousand times drivers from silabs. But pc still doesnt see it and board still flashing. So i try to change my usb wire to another one, and pc recognized it)) So if yr pc doesnt see com port try ty change the wire.

degustator, 2018-04-01:
Thanks to sshocked comment! Wire was the reason. Funniest moment  is that I expierenced this problem before and forgot. Lots of wires, that are supplied with devices for charging, aren't complete USB wires, there are probably ony 2 power leads inside

phani5016, 2018-04-05:
@sshocked ; thankyou... it worked perfectly.. just changed the cable.... 

can i please know why  the cable makes  a difference? does it have anything to do with the length and impedance associated with it?

ofSanjay, 2018-05-01:
thanks for these conversations. I try to change the cable and it works.

arbelaju, 2018-05-17:
I tried to change my wire but it didn't work. So I solved installing this:

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

SeppPenner, 2018-07-09:
I had to install the driver for the NodeMCU manually: https://github.com/nodemcu/nodemcu-devkit/tree/master/Drivers. Afterwards, it worked.

chronoss09, 2018-07-12:
Thanks @SeppPenner

WayneRiesterer, 2018-08-04:
@sshocked, nice work. It's been a while since I've dealt with hardware and forgot about this. I've faced this issue quite a few times before and keep forgetting that some cables don't offer a data connection. It's now working fine...on with the show :)

EOM, 2018-11-17:
Same problem in the end was the change that does not have to pass data, was a cable for cellular charge, so I lost an hour in search of the solution.
**He passed**
1 - Install the USB-Serial Chip driver: CP2102 is CP210x (https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers)
2 - Connect USB cable with Data cable and ready
3 - Ready we enter Arduino and you have the COMx serial

JPLoganlant, 2018-11-25:
Try a different cable

cahbagus07, 2019-02-17:
if you use windows 10, use the windows 8 driver

Harshac2, 2019-02-28:
> Hello I purchased a esp8266 NodeMCU and I connected the micro USB into the NodeMCU and connected via USB to my computer, which is running Windows 10 x64 bit. Nothing is showing up. However, the blue [...]

same in case of my node mcu

cahbagus07, 2019-02-28:
You can install the driver nodemcu

jiashanzh, 2019-03-15:
Same problem.Finally, i find my micro usb-cable ,can only use for charge, cannot work well

SeppPenner, 2019-03-16:
@jiashanzh This is also a quite common problem. Broken cables or cables that are usable for loading only...

PabloVortex, 2019-03-29:
Same problem here. Resolved by changing wire. Thx

MickPBduece, 2019-04-04:
> "I had same problem but I solved it. My board flashed only flashed once(it all about I think what my usb wire works) but, pc still doesnt recognize it. So i installed like 1 thousand times drivers [...]

Some of ports work for flashing an some do not.  The ones that don't will power a device and copy files from my phone.   Maybe it's how the driver is applied?   I'd be curious if anyone has an idea why that would be.

pixelartartist, 2019-04-15:
I have the same problem. I have downloaded ALL the drives, but they always say "Unable to find a device designed for this driver"
I have switch my Micro USB cables with the HUNDREDS I have at home and none of them work. It would be nice if they all were for charging, but most likely it's my pc.

Now, my pc does recognize that my a device is plugged in, but calls it Silicon labs CP210x USB to UART Bridge (COM3)

My Arduino IDE does not recognize this however. Any advice?

Harshac2, 2019-04-15:
I finally  figured the problem out.
As it  my arduino editor softwares inability to find and access the
drivers.I just changed my arduino version 1.6.4 to 1.8.8 then the problem
was solved . With that I have done my mini project using esp8266.

Harshac2, 2019-04-15:
If ur problem still not solved by turning from one arduino version to
another,then try it in another version .When  I struck with this problem I
checked 4-5 versions to solve the problem

pixelartartist, 2019-04-15:
I have gone through versions 1.8.9-1.8.1
You sure this is will work?
Thanks for the advice though!

Harshac2, 2019-04-15:
It worked for me in 1.8.8 after trying four versions

pixelartartist, 2019-04-16:
Thanks @Harshac2, I seriously appreciate your help :)
Well, I look at the serial monitor and see that the module is scanning for networks, but not connecting to anything, it just continues to scan and nothing else WHILE telling me that there was an error compiling for the board.

Harshac2, 2019-04-16:
Does it showing which port your node mcu is  connected?

pixelartartist, 2019-04-16:
Yes but only in the Device Manager on my pc, not the IDE
I can connect my arduino Uno  and I will see on the IDE which port it is connected too, but not the ESP8266

Harshac2, 2019-04-16:
Did you extracted all the driver in arduino files or not?

Harshac2, 2019-04-16:
If your esp8266 still not connected then try with another arduino version
as l did like that

CHEEKATLAPRADEEP-MSFT, 2019-05-24:
I have changed the usb cable, which worked...!

jlar0che, 2019-06-20:
Same here - changing my USB cable solved the problem!

TanmayNagdeve, 2019-09-29:
Leaving...
Hard resetting via RTS pin..

And nothing happens !!

oldandslow1, 2019-10-06:
Also having similar issues of it not being detected by the pc, tried a couple of cables both of which work with Arduinos. 
Drivers are installed and work with other things (arduinos dispark).

I found if I press rts it will show for a second then the go off.
If I press rts then flash it will remain visible and I can upload blink sketch, it then starts to reboot and disconnect over and over.

last line is the same as the above post
Leaving...
Hard resetting via RTS pin..

It's not really possible to test other sketches as I can't access serial monitor because it's disconnected.

oldandslow1, 2019-10-09:
Final update for anyone interested, the I believe the issue was lack of power.
I tried several cables all of which worked with arduinos and eSP32s but all failed with ESP8266.
They were all 1 meter long cables  so I purchased a shorter cable 0.5 meter and the ESP8266 works perfectly.

I did try the other cables again without success.

SeppPenner, 2019-10-10:
@oldandslow1 I can't imagine that this issues corresponds to the length of the cables... Do both cables have a data wire? (Some USB cables only support power, not data...)

Nevertheless, checking the cables is always my first check now (I had this issue before with a broken USB cable and nearly needed half a day to figure that out)...

oldandslow1, 2019-10-10:
> @oldandslow1 I can't imagine that this issues corresponds to the length of the cables... Do both cables have a data wire? (Some USB cables only support power, not data...)
>
> Nevertheless, checking [...]

The other cables came with various devices but I can use them to program other Arduinos and ESP32's without issues  and as mentioned they will program the ESP8266 but only after fiddling with the buttons (I don't need to do this with new cable)
Even after pressing the buttons  it wasn't working correctly, from what I read the ESP8266 draws quite a bit of current at start up and if it can't get enough it goes into a loop of trying to restart.

SeppPenner, 2019-10-10:
> The other cables came with various devices but I can use them to program other Arduinos and ESP32's without issues and as mentioned they will program the ESP8266 but only after fiddling with the buttons (I don't need to do this with new cable)
Even after pressing the buttons it wasn't working correctly, from what I read the ESP8266 draws quite a bit of current at start up and if it can't get enough it goes into a loop of trying to restart.

Ok, really strange...

pugrustler, 2019-10-13:
Had the same problem not connecting to pc. Tried other wires, install uninstall drivers, tried another cable. Nothing. Then I tried a good thick USB cable and it came to life. Don't use cheap USB cables.

Googlewasmyidea, 2019-11-23:
I cannot believe this. The actual reason was because of the wire... Thanks @sshocked !

boraaslan, 2019-11-30:
Some of usb wire is so dud. Please change the wire and try again. It will work %99.

jpreiss, 2020-01-10:
For what it's worth, I was able to flash `blink` on a 2013 rMBP running 10.14 Mojave with no issues. I used the driver from https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers .

parzibyte, 2020-02-08:
> I had to install the driver for the NodeMCU manually: https://github.com/nodemcu/nodemcu-devkit/tree/master/Drivers. Afterwards, it worked.

Yes, this is working. Thanks :)

jeevansahu, 2020-04-19:
the blue led is not blinking in nodemcu(generic ) i install their driver also but its not working what would i do?

disneyn, 2020-04-28:
> I had same problem but I solved it. My board flashed only flashed once(it all about I think what my usb wire works) but, pc still doesnt recognize it. So i installed like 1 thousand times drivers [...]

I just wanted to say THANKS to user sshocked for helping me with this issue, it worked perfect just to swap to another cable! //Hans

MickPBduece, 2020-04-28:
Maybe you did already but I marked a cable as being a good one after this
experience.

sahilshaikh89, 2020-05-16:
Wire was the problem for me. Glad I found the comment from sshocked.

tinashekadiki, 2020-05-30:
Installed drivers and it worked, https://github.com/nodemcu/nodemcu-devkit/tree/master/Drivers

Langfo, 2020-06-08:
After a couple of months of trying different configurations to get my ESP8266 to work such as laptops running Windows, VM's running Linux, a Macbook and a Raspberry PI.  Then trying to install various drivers on Windows, OSX and Linux.  Then swapping in and out over 10 USB cables.  In the end what worked was finally finding a USB cable that was thicker (meaning this had the data wires connected and wasn't just a charging cable) as well as installing the nodemcu drivers for Windows as mentioned and linked throughout this comment thread.  You will know a data USB cable when you see it, it will look much thicker (see image)
[image]

dhajnes, 2020-07-08:
Hi,
I had a somewhat similar problem. I bought some spinoff ESP8266 NodeMCU, which is not really the NodeMCU, but a piece of hardware called "HW-628 V1.1".
I installed ALL THE DRIVERS on the internet (really, like at least 15 different) and then I read somewhere about those spinoffs not coming with any software installed on them, and thats apparentely an issue.

First, you need to get drivers for your UART/USB converter device built in your deck next to the USB (either CH340G - a rectangle or a CP2102 - a square).

For mine (CH340G) i found some here: https://sparks.gogo.co.nz/ch340.html . Downloaded .zip, installed .exe, installed the needed .inf file and then my wifimodule was finally recognised by my PC (Win 7 - yeah i know, its 2020 and i still got win7, so waht?).

Wish you luck!

brother27, 2020-07-21:
Thanks for this post. I will have to go through my cables to find and mark the good ones.

technoo10201, 2020-07-22:
> I had same problem but I solved it. My board flashed only flashed once(it all about I think what my usb wire works) but, pc still doesnt recognize it. So i installed like 1 thousand times drivers [...]

Thanks you so much ! It worked for me.... Incredible 👍

Al3x3i, 2020-07-25:
I had the same issue, luckily have found a cable that works. Some of them worked differently:
Cable:1 -> didn't do anything, just nothing, no LED
Cable 2 -> LED was blinking on the board but I could not connect/upload data. Did not appear in `sudo ls /dev/tty*`
Cable 3 -> worked fine, the USB cable was from Samsung S4 mini.

All cables which I have tried had the same thickness, only they had different physical "pinout". I tried on Windows10 and Ubuntu.

Arjen997, 2020-08-01:
hi guys!

I also have some little troubles with this little device.
when i plug it into my computer, i can hear the sound, but...
my system does not recognize the device.

when i plug it in, the blue light will turn on and thats it.

an sound and an light.
i tried several things like installing the new driver, changing cables and at device management i turned hidden devices on.

i wonder if someone has any idea what to do.

WiliTest, 2020-08-26:
When one nodemcu isn't detected by the PC (but the others are), is it necessarily dead(fried or bricked)?

brother27, 2020-08-27:
These are small components. I use a magnifying glass in order to see anything. Glad you found the issue

mikyjazz, 2020-09-01:
Just to have my say, after two days of testing and swearing I discovered that three (3) usb cables I was trying were "power only" !!! I have successfully used the cable from my Asus tablet!

fkrnglc, 2020-10-19:
> I had to install the driver for the NodeMCU manually: https://github.com/nodemcu/nodemcu-devkit/tree/master/Drivers. Afterwards, it worked.

You are a life saver man. It solved my problem instantly.

[36 more replies]