esp8266/Arduino #3552: Can I use ESP8266WiFi library with ESP8266 ESP-01 Serial Port WIFI Transceiver module....? If yes than how should I connect esp01 to my arduino mega board?

maddy-jp, 2017-08-27:
Hello all,

I am new to arduino and wifi modules...........what I am trying to make is a wifi client and I want to make this using MEGA 2560 along with ESP8266..........This is a system where ESP01 should send data to database by making http GET requests.......so my question is Can I use this library directly by connecting ESP01 to the serial port or I need some other type of connection? the reason why I am asking this is even in the examples I didn't saw any mention of SoftwareSerial library or initialization of Rx/Tx pins of arduino to communicate with ESP01. Please help me.

suculent, 2017-08-27:
The SoftwareSerial library has examples inside - just install it and you'll find them in File menu. 

I've tried it with Bluetooth LE interfaces and works quite well. 

You may need serial TTL converter between 5V Arduino and 3,3V ESP. Connect any free digital pins through that to Arduino and declare those to be RX&TX on softwareserial initialization. 

Odesláno z iPhonu

vicnevicne, 2017-08-27:
Please note though that if you only intend to use the ESP8266 as a gateway to perform HTTP calls based on serial requests, you might as well NOT use this library at all and keep the stock "AT" firmware that most ESP-01 come programmed with.
See for example this instructable: http://www.instructables.com/id/Arduino-Esp8266-Post-Data-to-Website/
This has the great advantage of keeping your code in one place (the Arduino) and just using the ESP-01 with a "standard", widely used firmware.
Now if you want to write code to run on the ESP8266 using the familiar Arduino IDE, this library (actually, "core" is the appropriate term) is the solution... but then why use an Arduino in the first place ? ;-)

Anyway, this issue tracker is meant to discuss issues inside the ESP8266 core and not the right place to ask usage questions about either scenario, so I would kindly suggest that you close this issue and direct your questions to the esp8266.com community forums...

maddy-jp, 2017-08-28:
Thank you suculent and vicnevicne got the point. I am new to this and had no idea when I posted sorry for that. Sure I will look in to the AT commands and How to use those along with SoftwareSerial library.

closed 2017-08-28 by maddy-jp