esp8266/Arduino #3553: Using OTA Updates via Arduino Command Line

odessa2, 2017-08-27:
Is there any possibilty to use Arduino Command Line to flash a sketch via OTA? I can build and upload sketches using command line, but how I have not figured out how to use OTA. Is this even possible? 

Current Code for wired flashing:
`arduino --board esp8266:esp8266:d1_mini --port /dev/ttyUSB0 --upload tmp_build_dir/sketch.ino --verbose --pref build.flash_ld=eagle.flash.4m.ld`

I cannot find anything in the documentation.

reaper7, 2017-08-28:
espota.py -i ESP_IP_ADDR -f SKETCH.INO

tttapa, 2017-08-28:
I use the following Bash script. You first have to compile it the sketch. The script will scan for Arduino OTA services on the network. When that's finished, enter the path to the binary and the OTA password. It will upload the binary to all devices simultaneously.
```bash
echo "Scanning for Arduino OTA devices on the network ..."
ip_arr=($(((avahi-browse _arduino._tcp --resolve --parsable --terminate) 2>/dev/null | grep -F "=;") | cut -d\; -f8))
if [ ${#ip_arr[@]} == 0 ]
then
    echo "No devices found"
    exit
fi
echo "IP addresses found: "
printf '\t%s\n' "${ip_arr[@]}"

read -p "Path to binary: " path
read -p "Password: " password

read -p "Press [ENTER] to start OTA"
echo    "--------------------------"

for ip in ${ip_arr[@]}
do
    python2 /home/$USER/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/espota.py -i $ip -p 8266 --auth="$password" -f "$path" 2> /dev/null && echo -e "Success:\t$ip" || echo -e "Fail:   \t$ip" &
done
wait
```
Pieter

closed 2017-12-27 by igrr