esp8266/Arduino #3557: Does EEPROM data collites with SPIFFS

student1704, 2017-08-29:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	?2.1.0-rc2?

### Description

Problem description
 I use EEPROM.h to store some 4 bytes.  During power on I write and read the data. That confirms, the code ist ok.  But after power down the result is different. I use 3 M Spiffs.
 

### Settings in IDE

Module:  			NodeMCU 1.1 ESP12E
Flash Size: 		4MB/3m for SPIFFs
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz?
Upload Using:		/ SERIAL
Reset Method:		nodemcu?

student1704, 2017-08-29:
I use EEPROM.h to store some 4 bytes. During power on I write and read the data. That confirms, the code ist ok. But after power down the result is different. I use 3 M Spiffs.

pieman64, 2017-08-29:
@student1704  no it doesn't. Problem is normally bad code provided by the ESP user. Create the shortest possible sketch that demonstrates your problem and paste it here. Someone might then debug it for you.